Asuswrt-Merlin/asuswrt-merlin.legacy #1272: Change behavior of WPS button to toggle radio wifi in ASUS RT-AC66U B1

srgcdev, 2017-03-05:
Hello,

The firmware of ASUS RT-AC66U B1 router is the same that ASUS RT-AC68U.

ASUS RT-AC68U has a physical button to toggle the radio wifi but ASUS RT-AC66U B1 hasn't.

So it would be necessary to add an option to change the behavior of WPS button to toggle radio wifi for ASUS RT-AC66U B1.

Thanks.

RMerl, 2017-03-06:
Duplicate #1021

closed 2017-03-06 by RMerl

srgcdev, 2017-03-06:
@RMerl, #1021 describes a problem with LED's. The problem I describe here is related to have a physical button to toggle radio wifi changing the behavior of WPS button. So i think is not duplicated.
Thanks.

RMerl, 2017-03-07:
I posted in that other issue that I had no intention of hardcoding a specific feature, because there's always someone else asking for yet another feature.  Instead, the plan outlined in that other issue is to make it run any user script written by the user, which would allow the button to be used for any functionality desired by the user.  However that's a low priority, and I haven't found the time to implement this yet, so there's no ETA.