Asuswrt-Merlin/asuswrt-merlin.legacy #1278: Add "Route Modem IP" as per the Tomato feature

imavroukakis, 2017-03-10:
Tomato has a feature where by it allows you to access the LAN interface of your modem when it is in bridge mode. They call this "Route Modem IP" and it would be very useful if this feature could be added. For reference, see this discussion here : http://tomatousb.org/forum/t-290560/route-modem-ip-option

imavroukakis, 2017-03-10:
I'm aware of the solution provided at https://www.snbforums.com/threads/accessing-modem-via-n66u-and-wps-hack.8823/ however since there's an easier way, it should at least be considered :)

RMerl, 2017-03-15:
I want to avoid the feature-creep factor of other third party firmware.  That's one of those features I'd rather not add, due to limited usefulness, and the ability of implement it manually through scripting.

closed 2017-03-15 by RMerl

imavroukakis, 2017-03-15:
I agree wrt to feature creep. However, this particular feature is a very common use case, which I feel would benefit from a click and do-it solution, instead of scripts. Scripts have the added disadvantage of having to be backed up (since they live in the JFFS partition), in addition to configuration, making this a needless two step dance for the sake of a couple of scripts.