Asuswrt-Merlin/asuswrt-merlin.legacy #1284: Radio module not enabled after power on

muibusan, 2017-03-14:
RT-N66U. Issue firstly introduced with 380.65, also affects 380.65_2.

As mentioned in the change log a reboot bug had been fixed before:
> FIXED: Some models (N66/AC66/AC5300) would reboot 3 times
>             if one of the radios was found disabled by the user
>             while booting (Asus bug).

However, instead of the 3 reboots the router now simply doesn't enable the radio module after power on. Internet connection for ethernet based devices works as expected, just the radio module doesn't get switched on on boot.

This issue occurs from time to time, not at each system start. Maybe a timing issue. And it only seems to happen when the device was unpowered before. I only have the 5GHz-band-module enabled.

RMerl, 2017-03-15:
Sounds like a hardware issue then since Asus only does that for specific models.

Does it always work after just one reboot?  If so, I could set it to only reboot once instead of up to three times, as a compromise.

muibusan, 2017-03-18:
Oh, a hardware issue, I see.

Well, it's hard to tell for me. Maybe booting once would be a compromise. Currently I estimate it's around 50:50 to see the module working after a power on of the router. Maybe a reboot helps, let's test it.

closed 2017-04-10 by RMerl