Asuswrt-Merlin/asuswrt-merlin.legacy #1286: Wrap postconf scripts with timeout

imavroukakis, 2017-03-15:
According to https://github.com/RMerl/asuswrt-merlin/wiki/Custom-config-files 

> Note that postconf scripts are blocking the firmware while they run, to ensure the service only gets started once the script is done. Make sure those scripts do exit properly, or the router will be stuck during boot, requiring a factory default reset to recover it.

It might be useful to wrap execution of those scripts with something like this http://www.bashcookbook.com/bashinfo/source/bash-4.0/examples/scripts/timeout3

RMerl, 2017-03-15:
Not sure this would work, as Asuswrt uses ash, not bash.

The firmware has a function that allows to launch an external process with a timeout, however at the time I implemented postconf scripts, this function didn't work properly.

imavroukakis, 2017-03-15:
doable. Here's a related script https://gist.github.com/Juul/d8408750ef1dd20f9ae60d2d905dbf47

demonbane, 2017-04-03:
The "related" script isn't actually related in any way. This just implements a delay before running a process which is vastly different than killing an already-running process.

With that said though, I think the bash example (which is actually quite elegant IMO) wouldn't be too difficult to port to ash.

@RMerl I'd be happy to take it on if you think it would be a useful addition.

imavroukakis, 2017-04-04:
Relevant in the context of the delay, not what it ultimately does. Glad you found the script elegant

imavroukakis, 2017-04-05:
@RMerl you mentioned a firmware function that controls timeouts, would you point me towards it? Thanks!