Asuswrt-Merlin/asuswrt-merlin.legacy #1289: Feature Request: Add Let's Encrypt Support

jlambright, 2017-03-16:
I know this is a bit far fetched, but with the improvements to Let's Encrypt's certbot, it seems less so. Being able to automate SSL certs, even via command line, would be a wonderful benefit to the firmware. 

[Certbot Source](https://github.com/certbot/certbot)
[Certbot Packaging Guide](https://certbot.eff.org/docs/packaging.html)

RMerl, 2017-03-17:
Aside from the fact that Python would add 5-10 MB to the firmware just for certbot, Let's Encrypt is also tricky to implement as it requires your router to be exposed to the Internet, and it also requires you to own a domain name.  At this time there are no plans to implement Let's Encrypt support.

closed 2017-03-17 by RMerl

chaoscreater, 2017-06-19:
Technically you don't need a domain name.

I've setup a LetsEncrypt cert on my router using just the blah.asuscomm.com DDNS. Then I can just use a reverse proxy like nginx to route 443 traffic to whatever is on the backend - e.g. https://blah.asuscomm.com:9191

At least, 443 traffic on the DDNS would be validated by the LetsEncrypt cert.

To set up the LetsEncrypt cert manually, get a LetsEncrypt cert first and then follow this:
https://www.rickygao.com.au/blog/how-to-upload-ssl-certificate-to-asus-router/

yjo, 2017-09-28:
@RMerl I'd definitely encourage you to re-consider this request - we should really be encouraging people to avoid transmitting sensitive data (e.g. router passwords & auth cookies) over HTTP, even over a LAN given the huge attack surface within the network of a modern 'smart' home.  And if people are to use HTTPS, then the downsides of having to manually check and install certificates on client devices (or, more likely, clicking through browser warnings and accepting an untrusted certificate) are pretty huge.

To address your concerns:
* I definitely agree that the Python Certbot would be overkill -- I'd recommend [acme.sh](https://github.com/Neilpang/acme.sh) instead.  To make this work we'd probably need a `/www/.well-known/acme-challenge` link to a directory on the tmpfs so that the script can create the ACME challenge files
* As an alternative to opening up admin HTTP(S) access to the internet acme.sh could verify domain control using its 'standalone' mode instead, though this adds a dependency on socat (<1MB) and would probably require some trickery with iptables or something so that external requests to port 80 can access acme.sh instead of the router's HTTP interface
* Yes, users would need a domain name (or names) that point at their router, but as @chaoscreater points out this could just be the free asuscomm DDNS name.

RMerl, 2017-09-28:
Let's Encrypt support is something Asus is working on their end.  No idea if that project is still undergoing or not, but I still have no intention of devoting time on this personally - that development time (which is already scarce) is better devoted elsewhere.

RMerl, 2017-09-28:
I'm already less than impressed by the whole thing, Asus's current implementation involves having a second, dedicated lighttpd instance to handle Let's Encrypt - a pretty big waste of flash space, as some models are already tight on available flash.