Asuswrt-Merlin/asuswrt-merlin.legacy #1295: export vnp client settings as .ovpn

mrdeeks, 2017-03-18:
Hi,

Feature request for an export of vpn client settings as .ovpn file. I'd like to think this makes sense since we can already import ovpn files. For me it would help a lot when tweaking / setting up new vpn connections.

Cheers!

ntam, 2017-03-19:
That already exist when you configure the openvpn server

RMerl, 2017-03-27:
As mentionned, that has been present in the firmware for well over a year already.

closed 2017-03-27 by RMerl

mrdeeks, 2017-03-27:
Thanks for your responses so far. I finally had time to followup, I have seen the export settings for the VPN server but what I meant is having an export setting on the OpenVPN client tab. There is an import already but in order to quickly and flawlessly to replicate client 1 i would love to export its settings and import these as a basis for client 2.

Cheers,
Deeks

AdeHarrison, 2017-10-31:
Eric, sorry if I'm being a bit dim here but whereabouts is this option? - I too would love to save a VPN client settings and use them as a basis to import for another vpn client - would save lots of messing about and typos!

RMerl, 2017-10-31:
You cannot export an existing client config, only a server config.  It's the "Export" button that shows on the OpenVPN Server page.

AdeHarrison, 2017-11-02:
thanks for that, thought I was going mad for a moment :-)

maybe a future improvement? would be so useful mate!