Asuswrt-Merlin/asuswrt-merlin.legacy #1299: DLNA stopped on new file

th3nu11, 2017-03-21:
I have download manager and DLNA server enabled and both use "NEW_FILES" folder.

When download manager completes a file download (so it copies the new file in "NEW_FILES" folder) DNLA server is stopped.

I have to restart dlna server. (it's enough go in DLNA admin page and click on "apply" button).

Thanks for attention and thanks for your great job.

PhilHannent, 2017-05-13:
I can confirm that also I have this issue. I am not using the download manager, I am just transferring files via the smb network share.

I had enabled the new status page, that worked initially. But coming back to my TV, I found the DLNA server was no longer accessible and that the status page timed out. Disabling the status page did not fix the problem.

I am using an RT-AC66U.

PhilHannent, 2017-05-23:
The latest 380.66 firmware fixed this issue for me.

th3nu11, 2017-05-25:
for me too, it seems fixed.

closed 2017-05-25 by th3nu11