Asuswrt-Merlin/asuswrt-merlin.legacy #1311: Block device's internet access from homepage client list

TJRef, 2017-04-03:
The app lets you easily block internet access for single devices. But there's no easy and fast way to do that within the web interface. There are however Icons that show whether or not the device has internet access.. Maybe you can make those clickable to disable/enable internet access for that device. Some devices don't need constant access to the internet (NAS) but sometimes they do. Would be nice to do that remotely when necessary.

Eric1911, 2017-04-15:
From the main "network map" page of the web interface, if you click on the round clients icon the "client status" column appears on the right side of the web interface (replacing the system status information). From there you can click on an individual client and a pop-up appears that allows you to block or unblock the client's internet access. Hope this helps.

TJRef, 2017-04-16:
oh wow, I missed that entirely. Never thought those were buttons... Thanks!

closed 2017-04-16 by TJRef

exbarboss, 2017-07-24:
Is there any option that should be switched on? I tried to do that for my iPad 2 and it still has internet connection.

latearrival, 2017-08-17:
+1 For not knowing those circles were clickable.  Have been using this firmware for years and never knew.  Perhaps it could be made more obvious they are clickable (especially as there's a button below that's screaming 'Click me!' (well, View List). I still think the intuitive place to have this functionality is in the View List, in the Internet Column. Where the globe icon shows access enabled/disabled, but it is not clickable.

postcert, 2017-12-30:
Doesn't look to be available option while in AP Mode?
Can't think of much else keeping my AC66U on 380.69 from using this feature (the client is connected to the AP's SSID)

Ended up just applying a mac rule on the edge modem/router though.

xiaomingmiao, 2018-01-07:
@Eric1911  AirPortection --  Parents Controls -- Time Scheduling.