Asuswrt-Merlin/asuswrt-merlin.legacy #1312: Console errors since 380.65_4

w84no1, 2017-04-03:
Uncaught TypeError: Cannot read property 'getElementsByTagName' of null
    at refreshStatus (state.js:2879)
    at XMLHttpRequest.xmlHttp.onreadystatechange (makeRequest.js:19)
refreshStatus @ state.js:2879
xmlHttp.onreadystatechange @ makeRequest.js:19
state.js:2879 Uncaught TypeError: Cannot read property 'getElementsByTagName' of null
    at refreshStatus (state.js:2879)
    at XMLHttpRequest.xmlHttp.onreadystatechange (makeRequest.js:19)
refreshStatus @ state.js:2879
xmlHttp.onreadystatechange @ makeRequest.js:19

imavroukakis, 2017-04-05:
I assuming you mean browser console. Any specific page? I can't seem to replicate it with mine.

On Chrome 57.0.2987.133

w84no1, 2017-04-05:
Yes, I am using the Chrome inspector. I reverted back to a previous version that did work without errors and it still has those errors on every page.  I think it might be my setup (usb modem tethered), so I am going to factory reset it and see if that fixes it.

closed 2017-09-09 by RMerl