Asuswrt-Merlin/asuswrt-merlin.legacy #1313: Possible VPN routing issue

imavroukakis, 2017-04-05:
I've noticed some odd behaviour when using a VPN coupled with IP based VPN routing policy. When the VPN goes down for any reason (manual or otherwise), all WAN access is lost. It's almost as if the routing tables are not cleared once the VPN connection has gone. Has anyone else noticed this?

RMerl, 2017-04-10:
Please use the support forums at SNBForums, as there are many of us using policy routing without any problem.  Something must be specific to your config.

closed 2017-04-10 by RMerl

imavroukakis, 2017-04-10:
@RMerl I don't think so. My config is as bog standard as it gets. I *do* however see some interesting errors in the logs, perhaps you'd like to take another look

```
Apr  7 15:53:06 openvpn-routing: Configuring policy rules for client 1
Apr  7 15:53:06 openvpn-routing: Removing rule 10101 from routing policy
Apr  7 15:53:06 openvpn-routing: Removing rule 10102 from routing policy
Apr  7 15:53:06 openvpn-routing: Tunnel down - VPN client access blocked
Apr  7 15:53:06 openvpn-routing: Adding route for 10.0.0.19 to 0.0.0.0 through VPN client 1
Apr  7 15:53:06 openvpn-routing: Adding route for 10.0.0.20 to 0.0.0.0 through VPN client 1
Apr  7 15:53:06 openvpn-routing: Completed routing policy configuration for client 1
Apr  7 15:53:06 openvpn[16741]: /usr/sbin/ip route del 10.200.0.1/32 metric 1
Apr  7 15:53:06 openvpn[16741]: ERROR: Linux route delete command failed: external program exited with error status: 2
Apr  7 15:53:06 openvpn[16741]: /usr/sbin/ip route del 5.x.x.x/32
Apr  7 15:53:06 openvpn[16741]: /usr/sbin/ip route del 0.0.0.0/1
Apr  7 15:53:06 openvpn[16741]: ERROR: Linux route delete command failed: external program exited with error status: 2
Apr  7 15:53:06 openvpn[16741]: /usr/sbin/ip route del 128.0.0.0/1
Apr  7 15:53:06 openvpn[16741]: ERROR: Linux route delete command failed: external program exited with error status: 2
Apr  7 15:53:06 openvpn[16741]: Closing TUN/TAP interface
Apr  7 15:53:06 openvpn[16741]: /usr/sbin/ip addr del dev tun11 local 10.200.22.18 peer 10.200.22.17
Apr  7 15:53:07 openvpn[16741]: SIGTERM[soft,ping-exit] received, process exiting
```