Asuswrt-Merlin/asuswrt-merlin.legacy #1315: Advanced_DHCP_Content Page not functioning correctly

tholken, 2017-04-09:
The Page Advanced_DHCP_Content.asp will not allow for an APPLY button to work. The surrounding navigation page is missing.  Any changes to the page will not be accepted or allow modifications. I have tried with Firefox, chrome, and IE, all will fail.

Firmware:380.65_4  running on a RT-AC3200

Steps to replicate:
Click LAN under "Advanced Settings"
Click the "DHCP Server" tab
Navigation page disappears
Attempt to select a MAC Address from the drop down with no action.

Source code has the VALUES set but the drop down does not populate.

Any changes to the page can be made, but the APPLY button does not function

madbuda, 2017-04-10:
I have the same issues on 
Firmware:380.63_2 running on a RT-AC66U

RMerl, 2017-05-10:
Chances are you have a LAN client that is defined with a quote in its hostname, breaking the Javascript code.

380.66 has some additional protection to better deal with these, give it a try.

madbuda, 2017-05-16:
Interesting... I was able to add a few manually and now it seems to work fine.

closed 2017-05-30 by RMerl

ddaversa, 2017-07-21:
I have the same problem. A quick look at the Chrome console revealed that the JavaScript error is due to a single quote ( ' ) in one of my DHCP hostnames, breaking the string assignment in line 62 of Advanced_DHCP_Content.asp.

A similar issue is visible if you click the "View List" button from the Network Map page (index.asp), the hostnames containing a single quote have their output broken as well.