Asuswrt-Merlin/asuswrt-merlin.legacy #1318: permission error with gendict.pl file

azdps, 2017-04-10:
I'm getting a permission error with the gendict.pl file during the build process. Changing the file permission to executable solved this problem. I cloned the asuswrt-merlin repository 3 times which resulted in the same permission error during each build process.

RMerl, 2017-04-18:
Works for everyone else, make sure you use a compatible filesystem.  Asuswrt cannot be build using a Windows filesystem for instance.

closed 2017-04-18 by RMerl

azdps, 2017-04-19:
If you set AUTODICT=n in the asuswrt-merlin/release/src-rt/target.mak  file and gendict.pl will fail to execute because of the permission issue. If you look at the file under linux, you can see it not set as executable which it should be. I doubt most users don't notice this.

I use linux mint to build asuswrt-merlin. I don't use windows at all.