Asuswrt-Merlin/asuswrt-merlin.legacy #1328: Port Forwarding does not work when Dos is enabled - ASUS RT-AC68U - Assuswrt-Merlin Firmware 380.65_4

rexutos, 2017-04-16:
**ASUS RT-AC68U - Assuswrt-Merlin Firmware 380.65_4 - Enable DoS protection Port Forwarding**

I have a router ASUS RT-AC68U with the firware Assuswrt-Merlin Firmware 380.65_4
I found an issue, which is:
when the function DoS protection is enabled the function Port Forwarding does not work (for more details, please see a piece of the error log below)
when the function DoS protection is disabled the function Port Forwarding works normally
Since the Ddos is so important nowadays, when do you think this bug will bo corrected?
Thank you in advance

--------------------------LOG-------------------
Apr 16 12:04:01 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=47524 DF PROTO=TCP SPT=62414 DPT=554 SEQ=1194850666 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78B54F0000000004020000) 
Apr 16 12:04:01 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=32870 DF PROTO=TCP SPT=62409 DPT=554 SEQ=997086450 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78B6C00000000004020000) 
Apr 16 12:04:01 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=9901 DF PROTO=TCP SPT=62411 DPT=554 SEQ=2152753471 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78B7170000000004020000) 
Apr 16 12:04:02 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=34846 DF PROTO=TCP SPT=62412 DPT=554 SEQ=3161315080 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78B7D20000000004020000) 
Apr 16 12:04:02 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=495 DF PROTO=TCP SPT=62415 DPT=554 SEQ=3037179573 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78B9E50000000004020000) 
Apr 16 12:04:02 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=25158 DF PROTO=TCP SPT=62409 DPT=554 SEQ=997086450 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78BAA80000000004020000) 
Apr 16 12:04:02 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=11122 DF PROTO=TCP SPT=62411 DPT=554 SEQ=2152753471 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78BB000000000004020000) MARK=0x42d400b9 
Apr 16 12:04:03 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=35995 DF PROTO=TCP SPT=62412 DPT=554 SEQ=3161315080 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78BBBB0000000004020000) 
Apr 16 12:04:03 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=49250 DF PROTO=TCP SPT=62416 DPT=554 SEQ=2596216706 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78BEE10000000004020000) 
Apr 16 12:04:03 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=47532 DF PROTO=TCP SPT=62411 DPT=554 SEQ=2152753471 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78BEE80000000004020000) 
Apr 16 12:04:04 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=55291 DF PROTO=TCP SPT=62412 DPT=554 SEQ=3161315080 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78BFA30000000004020000) 
Apr 16 12:04:04 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=45372 DF PROTO=TCP SPT=62416 DPT=554 SEQ=2596216706 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78C2CA0000000004020000) 
Apr 16 12:04:05 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=14877 DF PROTO=TCP SPT=62412 DPT=554 SEQ=3161315080 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78C38B0000000004020000) 
Apr 16 12:04:05 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=27491 DF PROTO=TCP SPT=62417 DPT=554 SEQ=1169616616 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78C3A40000000004020000) 
Apr 16 12:04:06 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=37642 DF PROTO=TCP SPT=62411 DPT=554 SEQ=2152753471 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78C6B80000000004020000) 
Apr 16 12:04:06 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=17603 DF PROTO=TCP SPT=62418 DPT=554 SEQ=3012515070 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78C77E0000000004020000) 
Apr 16 12:04:06 kernel: DROP IN=eth0 OUT=br0 SRC=x.x.x.x DST=y.y.y.y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=18590 DF PROTO=TCP SPT=62419 DPT=554 SEQ=3813839472 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A1D78C8380000000004020000) 
--------------------------LOG-------------------

RMerl, 2017-04-18:
I just tested it and it's working for me, forwarding the TCP port used for my ATA.

Try a recent alpha build to see if you get better luck:

https://www.mediafire.com/folder/bj94sbhrh7e49/Test_Builds

Also make sure you aren't DDoS'ing yourself in your test.

rexutos, 2017-04-19:
Thank you for your answer. 
I followed your suggestion of upgrading 380.66_alpha3 (RT-AC68U_380.66_alpha3-g93de9e0.trx)
I'm affraid the issue is not resolved yet
I'm forwarding from an IP camera (port 554 or port 80) 
When I try to access that same camera (with Dos on) from the net, either the page refresh is very very very slow or I get a timeout error
I think that I am not DoS'ing yourself because only my IP is visible in the cam's access logs (please see below)
Do you have any further comments or suggestions? that I would be grateful for
Best wishes

---------------------------- LOG ---------------------------------
Apr 19 19:45:38 kernel: ACCEPT IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=47111 DF PROTO=TCP SPT=39666 DPT=554 SEQ=3369232192 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3513550000000004020000) 
Apr 19 19:45:38 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=21553 DF PROTO=TCP SPT=39665 DPT=554 SEQ=403262617 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3513530000000004020000) 
Apr 19 19:45:38 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=3761 DF PROTO=TCP SPT=39667 DPT=554 SEQ=1111532172 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3514EB0000000004020000) 
Apr 19 19:45:38 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=29015 DF PROTO=TCP SPT=39668 DPT=554 SEQ=1830880544 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3515A00000000004020000) 
Apr 19 19:45:39 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=13140 DF PROTO=TCP SPT=39665 DPT=554 SEQ=403262617 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E35173B0000000004020000) 
Apr 19 19:45:39 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=8952 DF PROTO=TCP SPT=39664 DPT=554 SEQ=924078551 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E35173B0000000004020000) 
Apr 19 19:45:39 kernel: ACCEPT IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=3240 DF PROTO=TCP SPT=39667 DPT=554 SEQ=1111532172 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3518D40000000004020000) 
Apr 19 19:45:39 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=54332 DF PROTO=TCP SPT=39668 DPT=554 SEQ=1830880544 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3519880000000004020000) 
Apr 19 19:45:40 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=30004 DF PROTO=TCP SPT=39665 DPT=554 SEQ=403262617 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E351B240000000004020000) 
Apr 19 19:45:40 kernel: ACCEPT IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=30327 DF PROTO=TCP SPT=39664 DPT=554 SEQ=924078551 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E351B240000000004020000) 
Apr 19 19:45:40 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=27860 DF PROTO=TCP SPT=39669 DPT=554 SEQ=3091775780 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E351CE60000000004020000) 
Apr 19 19:45:40 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=53976 DF PROTO=TCP SPT=39668 DPT=554 SEQ=1830880544 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E351D700000000004020000) 
Apr 19 19:45:41 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=61276 DF PROTO=TCP SPT=39670 DPT=554 SEQ=3305745251 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E351DE90000000004020000) 
Apr 19 19:45:41 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=1189 DF PROTO=TCP SPT=39665 DPT=554 SEQ=403262617 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E351F0C0000000004020000) 
Apr 19 19:45:41 kernel: ACCEPT IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=59782 DF PROTO=TCP SPT=39669 DPT=554 SEQ=3091775780 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3520CF0000000004020000) 
Apr 19 19:45:41 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=24341 DF PROTO=TCP SPT=39668 DPT=554 SEQ=1830880544 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3521580000000004020000) 
Apr 19 19:45:42 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=4853 DF PROTO=TCP SPT=39670 DPT=554 SEQ=3305745251 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3521D10000000004020000) 
Apr 19 19:45:42 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=46306 DF PROTO=TCP SPT=39671 DPT=554 SEQ=3159701038 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E3521FB0000000004020000) 
Apr 19 19:45:42 kernel: DROP IN=eth0 OUT= MAC=2c:4d:54:49:87:60:00:01:5c:7d:bc:46:08:00 SRC=163.172.149.131 DST=81.84.16.149 LEN=40 TOS=0x00 PREC=0x00 TTL=239 ID=64208 PROTO=TCP SPT=46062 DPT=21320 SEQ=2390800653 ACK=0 WINDOW=1024 RES=0x00 SYN URGP=0 
[... cut ...]
Apr 19 19:45:53 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=6708 DF PROTO=TCP SPT=39424 DPT=554 SEQ=3523715479 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E354D8A0000000004020000) 
Apr 19 19:45:53 kernel: DROP IN=eth0 OUT=br0 SRC=X.X.X.X DST=Y.Y.Y.Y LEN=64 TOS=0x00 PREC=0x00 TTL=56 ID=5978 DF PROTO=TCP SPT=39679 DPT=554 SEQ=2472974722 ACK=0 WINDOW=65535 RES=0x00 SYN URGP=0 OPT (02040582010303050101080A3E354D8E0000000004020000) 
---------------------------- LOG ---------------------------------

RMerl, 2017-04-24:
I suggest you post on the support forums at SNBForums.  At this point I consider this more likely to be a configuration issue since I cannot reproduce it here.

closed 2017-04-24 by RMerl

rexutos, 2017-04-25:
thank you very much for your feedback
I find this firware very good but I hope the issue above will be resolved in a future release
Best,

jslay88, 2017-05-06:
Having the same issue. It appears it mainly effects ports where you are streaming. I can get some connectivity with DoS protection enabled, but anything where I need a solid stream, its trash.