Asuswrt-Merlin/asuswrt-merlin.legacy #1331: Support for FD_CLOEXEC and SOCK_CLOEXEC

daenney, 2017-04-20:
It seems that the ASUSWRT firmware is based of kernel 2.6.22 which lacks support for FD_CLOEXEC (introduced in 2.6.23) and SOCK_CLOEXEC (introduced in 2.6.27) that some programs have come to rely on (and not properly check if they can set and handle that situation).

I was wondering if it's at all feasible to build the firmware on top of a newer kernel version in the 2.6.x series and if there are any instructions around as to how I could go about in trying to do so?

RMerl, 2017-04-21:
The kernel is tied to the manufacturer's SDK and therefore cannot be changed.  Only the RT-N66U and RT-AC66U still use 2.6.22.19, all other supported models are on 2.6.36.

FD_CLOEXEC support is already there, but SOCK_CLOEXEC isn't.  This shouldn't be a problem since any program trying to use it would fail to build, as SOCK_CLOEXEC isn't defined in the kernel headers.

closed 2017-04-21 by RMerl

daenney, 2017-04-21:
> This shouldn't be a problem since any program trying to use it would fail to build, as SOCK_CLOEXEC isn't defined in the kernel headers.

It is, for Go. Since 1.8 it's possible to cross-compile for 32bit MIPS from any machine but as a consequence it'll break at runtime since it can't really detect the lack of SOCK_CLOEXEC at compile time.

I figured as much though. That's too bad. But if FD_CLOEXEC is already in there I might just try and lift the SOCK_CLOEXEC patch over and build my own to fix that.

Thanks for the info!