Asuswrt-Merlin/asuswrt-merlin.legacy #1334: Redirect loop after clicking on 2.4GHz AP name

dandv, 2017-04-22:
I'm trying to get this firmware to let me connect to the APs while [my router is in repeater mode](https://www.snbforums.com/threads/intermittently-unable-to-access-the-router-after-installing-latest-stable-build.38730/), to no avail.

When I click the 2.4GHz AP name at the top - which, by the way, does NOT appear clickable; please fix that - I get the redirect loop shown in the screencast below. If I click the 5GHz AP name, I get the settings page. Anyway, I can't connect to either of those APs ("Failed to obtain IP address")

[image]

dandv, 2017-04-23:
I've reverted to the ASUS firmware RT-AC68U_3.0.0.4_380_7378-g7a25649 and I can connect to both wireless APs, without having changed any settings in the router.

Has this Merlin build been tested on routers in repeater mode?

Also, if I click on the 2.4Ghz AP name, I still get the redirect loop, so this is a bug upstream.

Does ASUS backport fixes from this fork into their unified firmware? Would be great to mention this in the [About page](https://github.com/RMerl/asuswrt-merlin/wiki/About-Asuswrt) of the wiki.

RMerl, 2017-05-10:
Note that the fact that you can access this page through the SSID at the top is probably a bug in itself, as Asus removes the link to that parameter page on the left side menu.

The behaviour you see is probably caused by the fix that prevents bogus parameters from being shown on the wireless settings page if you just edited the Guest Network settings.  While it might be possible to re-engineer the fix to prevent this side-effect, this is a fairly low priority as you're not supposed to be able to reach this page in the first place.