Asuswrt-Merlin/asuswrt-merlin.legacy #1336: PPPoE doesn't work for some ISP

sknaumov, 2017-04-24:
PPPoE doesn't work for some ISP that use PPP local address instead of remote one as default GW. Solution is to add an option to select what PPP address should be used as GW (remote by default).

sknaumov, 2017-04-24:
[0001-ppp-add-an-option-to-use-local-PPP-address-as-defaul.patch.txt](https://github.com/RMerl/asuswrt-merlin/files/950705/0001-ppp-add-an-option-to-use-local-PPP-address-as-defaul.patch.txt)

themiron, 2017-04-24:
please provide pppd IPCP log:
1. set syslog log level to debug
2. add "debug" to additonal pppd's options

RMerl, 2017-08-06:
Which ISP would require this patch?

sknaumov, 2017-08-06:
It is a Russian ISP in the area I live: http://tinconet.ru/

goletsa, 2017-08-06:
I have had N66U with merlinwrt 378 on that ISP with no problems.
May be you have wrong settings?)

sknaumov, 2017-08-06:
No, settings are definitely OK. There was an ISP's router that worked with PPPOE credentials without extra settings. When I applied the same settings to ASUS RT-AC87U, I saw that ppp interface is up and running, but there is no Internet connection. Looking deeper I experimented with route table and Internet connection get up when I used ppp local IP address as default GW instead of remote one.

BTW, here are the logs:

> Aug  1 03:03:49 pppd[1290]: Plugin rp-pppoe.so loaded.
Aug  1 03:03:49 pppd[1290]: RP-PPPoE plugin version 3.11 compiled against pppd 2.4.7
Aug  1 03:03:49 pppd[1309]: pppd 2.4.7 started by admin, uid 0
Aug  1 03:03:49 pppd[1309]: PADS: Service-Name: ''
Aug  1 03:03:49 pppd[1309]: PPP session is 5609 (0x15e9)
Aug  1 03:03:49 pppd[1309]: Connected to f8:c0:01:18:5e:01 via interface eth0
Aug  1 03:03:49 pppd[1309]: using channel 2
Aug  1 03:03:49 pppd[1309]: Using interface ppp0
Aug  1 03:03:49 pppd[1309]: Connect: ppp0 <--> eth0
Aug  1 03:03:49 pppd[1309]: sent [LCP ConfReq id=0x1 <mru 1492> <magic 0x37ea4be3>]
Aug  1 03:03:49 pppd[1309]: rcvd [LCP ConfReq id=0xcb <mru 1492> <auth chap MD5> <magic 0x52e06469>]
Aug  1 03:03:49 pppd[1309]: sent [LCP ConfAck id=0xcb <mru 1492> <auth chap MD5> <magic 0x52e06469>]
Aug  1 03:03:49 pppd[1309]: rcvd [LCP ConfAck id=0x1 <mru 1492> <magic 0x37ea4be3>]
Aug  1 03:03:49 pppd[1309]: sent [LCP EchoReq id=0x0 magic=0x37ea4be3]
Aug  1 03:03:49 pppd[1309]: rcvd [CHAP Challenge id=0x57 <040813fbf5ace42267d4c1c5eea1be6190969b4f7dea6bc2e4092c5a89832b37>, name = "JUNOS"]
Aug  1 03:03:49 pppd[1309]: sent [CHAP Response id=0x57 <42d9c00169211014740cfaa11e695c9c>, name = "e4168"]
Aug  1 03:03:49 pppd[1309]: rcvd [LCP EchoRep id=0x0 magic=0x52e06469]
Aug  1 03:03:49 qharvestd[854]: qharvestd[error]: curl_easy_perform(): Couldn't resolve host 'maui.quantenna.com'
Aug  1 03:03:49 pppd[1309]: rcvd [CHAP Success id=0x57 ""]
Aug  1 03:03:49 pppd[1309]: CHAP authentication succeeded
Aug  1 03:03:49 pppd[1309]: CHAP authentication succeeded
Aug  1 03:03:49 pppd[1309]: peer from calling number F8:C0:01:18:5E:01 authorized
Aug  1 03:03:49 pppd[1309]: sent [IPCP ConfReq id=0x1 <addr 0.0.0.0> <ms-dns1 0.0.0.0> <ms-dns2 0.0.0.0>]
Aug  1 03:03:49 pppd[1309]: rcvd [IPCP ConfNak id=0x1 <addr 10.5.111.83> <ms-dns1 10.5.240.1> <ms-dns2 10.5.240.3>]
Aug  1 03:03:49 pppd[1309]: sent [IPCP ConfReq id=0x2 <addr 10.5.111.83> <ms-dns1 10.5.240.1> <ms-dns2 10.5.240.3>]
Aug  1 03:03:49 pppd[1309]: rcvd [IPCP ConfAck id=0x2 <addr 10.5.111.83> <ms-dns1 10.5.240.1> <ms-dns2 10.5.240.3>]
Aug  1 03:03:50 pppd[1309]: rcvd [IPCP ConfReq id=0xcb <addr 172.31.255.254>]
Aug  1 03:03:50 pppd[1309]: sent [IPCP ConfAck id=0xcb <addr 172.31.255.254>]
Aug  1 03:03:50 pppd[1309]: Script /tmp/ppp/ip-pre-up started (pid 1317)
Aug  1 03:03:50 pppd[1309]: Script /tmp/ppp/ip-pre-up finished (pid 1317), status = 0x0
**Aug  1 03:03:50 pppd[1309]: local  IP address 10.5.111.83
Aug  1 03:03:50 pppd[1309]: remote IP address 172.31.255.254
Aug  1 03:03:50 pppd[1309]: primary   DNS address 10.5.240.1
Aug  1 03:03:50 pppd[1309]: secondary DNS address 10.5.240.3**
Aug  1 03:03:50 pppd[1309]: Script /tmp/ppp/ip-up started (pid 1319)
Aug  1 03:03:50 rc_service: ip-up 1319:notify_rc start_firewall
Aug  1 03:03:50 dnsmasq[448]: read /etc/hosts - 5 addresses
Aug  1 03:03:50 dnsmasq[448]: using nameserver 10.5.240.1#53
Aug  1 03:03:50 dnsmasq[448]: using nameserver 10.5.240.3#53
Aug  1 03:03:51 start_nat_rules: apply the nat_rules(/tmp/nat_rules_ppp0_eth0)!
Aug  1 03:03:51 wan: finish adding multi routes
Aug  1 03:03:51 rc_service: ip-up 1319:notify_rc stop_upnp
Aug  1 03:03:51 rc_service: waitting "start_firewall" via ip-up ...
Aug  1 03:03:52 rc_service: ip-up 1319:notify_rc start_upnp
Aug  1 03:03:52 rc_service: waitting "stop_upnp" via ip-up ...
Aug  1 03:03:53 ntp: start NTP update
Aug  1 03:03:57 ntp: start NTP update
Aug  1 03:03:58 rc_service: ip-up 1319:notify_rc start_firewall
Aug  1 03:03:58 pppd[1309]: Script /tmp/ppp/ip-up finished (pid 1319), status = 0x0

As you can see, advertised DNS servers are in PPP local subnet.

goletsa, 2017-08-06:
> As you can see, advertised DNS servers are in PPP local subnet.
No, local subnet have /32 mask and doesn't overlap with dns subnet.

themiron, 2017-08-06:
well, guys, there's unclear in what subnet PPP DNSes do belong.
according log ppp log, there will be

172.31.255.254/32 via ppp src 10.5.111.83
default via 172.31.255.254/32 via ppp src 10.5.111.83

route to 10.5.240.1/32 and to 10.5.240.3/32 will be either via ppp, or via anything else, if there's more narrow route (via MAN for example).

sknaumov, 2017-08-06:
Anyway it is clear that while default route was via 172.31.255.254, nothing worked, and when it became 10.5.111.83 (patch applied, "local" setting selected), everything works flawlessly for more than 3 months.

goletsa, 2017-08-06:
`# ip a
1: lo: <LOOPBACK,MULTICAST,UP,10000> mtu 16436 qdisc noqueue
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 brd 127.255.255.255 scope host lo
2: eth0: <BROADCAST,MULTICAST,UP,10000> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:26:18:4c:ba:7f brd ff:ff:ff:ff:ff:ff
3: eth1: <BROADCAST,MULTICAST,ALLMULTI,UP,10000> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 00:26:18:4c:ba:7f brd ff:ff:ff:ff:ff:ff
4: vlan1@eth0: <BROADCAST,MULTICAST,ALLMULTI,UP,10000> mtu 1500 qdisc noqueue
    link/ether 00:26:18:4c:ba:7f brd ff:ff:ff:ff:ff:ff
6: vlan11@eth0: <BROADCAST,MULTICAST,UP,10000> mtu 1500 qdisc noqueue
    link/ether 00:26:18:4c:ba:7f brd ff:ff:ff:ff:ff:ff
    inet 10.0.161.59/22 brd 10.0.163.255 scope global vlan11
9: br0: <BROADCAST,MULTICAST,ALLMULTI,UP,10000> mtu 1500 qdisc noqueue
    link/ether 00:26:18:4c:ba:7f brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br0
10: ppp0: <POINTOPOINT,MULTICAST,UP,10000> mtu 1492 qdisc pfifo_fast qlen 3
    link/ppp
    inet 10.5.111.84 peer 172.31.255.254/32 brd 10.5.111.84 scope global ppp0
`

`
10.5.240.1 via 10.0.160.2 dev vlan11  metric 1
172.31.255.254 dev ppp0  proto kernel  scope link
8.8.8.8 via 10.0.160.2 dev vlan11  metric 1
192.168.1.0/24 dev br0  proto kernel  scope link  src 192.168.1.1
10.0.160.0/22 dev vlan11  proto kernel  scope link  src 10.0.161.59
127.0.0.0/8 dev lo  scope link
default via 172.31.255.254 dev ppp0
default via 10.0.160.2 dev vlan11  metric 1
`
`# nslookup mail.ru
Server:    10.5.240.1
Address 1: 10.5.240.1

Name:      mail.ru
Address 1: 2a00:1148:db00:0:b0b0::1 mail.ru
Address 2: 94.100.180.200 mail.ru
Address 3: 217.69.139.201 mail.ru
Address 4: 217.69.139.200 mail.ru
Address 5: 94.100.180.201 mail.ru

# nslookup mail.ru 10.5.240.1
Server:    10.5.240.1
Address 1: 10.5.240.1

Name:      mail.ru
Address 1: 2a00:1148:db00:0:b0b0::1 mail.ru
Address 2: 94.100.180.201 mail.ru
Address 3: 94.100.180.200 mail.ru
Address 4: 217.69.139.201 mail.ru
Address 5: 217.69.139.200 mail.ru
admin@(none):/# nslookup mail.ru 10.5.240.3
Server:    10.5.240.3
Address 1: 10.5.240.3

# nslookup mail.ru 10.5.240.3
Server:    10.5.240.3
Address 1: 10.5.240.3

Name:      mail.ru
Address 1: 2a00:1148:db00:0:b0b0::1 mail.ru
Address 2: 217.69.139.200 mail.ru
Address 3: 217.69.139.201 mail.ru
Address 4: 94.100.180.200 mail.ru
Address 5: 94.100.180.201 mail.ru
`

I find one strange route via MAN, but anywhere - dns reply correct.
You can simply disable VPN+DHCP to ignore this.

All other traffic  work via default remote
`# traceroute -n 10.5.240.3
traceroute to 10.5.240.3 (10.5.240.3), 30 hops max, 38 byte packets
 1  172.30.0.2  1.152 ms  0.748 ms  0.619 ms
 2  172.30.2.36  0.758 ms  0.634 ms  0.726 ms
 3  10.5.240.3  0.677 ms  1.110 ms  0.660 ms
`