Asuswrt-Merlin/asuswrt-merlin.legacy #1343: Merlin Firmware leaves open ports in its default state (with

matt-gb, 2017-04-27:
I've configured my Merlin with OpenVPN (very nice integration btw!)

So I thought next...I should at least do a quick port-scan on the puppy to see whats open..
OpenVPN : UDP port 1194 and another on TCP 443)

Plug in my iphone to my PC, disable wifi, run up a VPN (I dont want to be flagged as a bad guy!)

nmap at my IP and what do I see...

PORT    STATE SERVICE
21/tcp  open  ftp
25/tcp  open  smtp
80/tcp  open  http
443/tcp open  https

So I figure OpenDNS is 443 - but leaving open 21 (FTP), 25 (SMTP) and 80 (http) - by default..??

I tried telnetting to those ports, and it didnt show me much - however surely we should have 21, 25 and 80 disabled by default on the WAN side..?

many thanks!
M

Syst3mSh0ck, 2017-04-29:
You are testing the VPN provider endpoint, not your local WAN IP connectivity. Most VPN providers will leave these ports open by default so nothing to worry about.

RMerl, 2017-05-04:
As pointed out, these ports aren't open by the firmware, but by your configuration.

By default, out of the box, there are zero ports open to the WAN.

closed 2017-05-04 by RMerl