Asuswrt-Merlin/asuswrt-merlin.legacy #1348: AsusWRT sends network traffic data to Trend Micro if certain features enabled

diimdeep, 2017-05-05:
Is this true about asuswrt-merlin ?

- https://ctrl.blog/entry/review-asuswrt#section-asuswrt-privacy
- https://news.ycombinator.com/item?id=14254766

> There is, however, a catch in using these services. ASUSWRT will collect and transmit data about which websites you visit to Trend Micro, if you use any of the following features in ASUSWRT:

- Apps/traffic Analysis
- Bandwidth Monitor
- Network Analyzer
- Network Protection (AiProtection), blocks known malware domains
- Parental Controls, including time scheduling
- Quality-of-Service
- Web History

RMerl, 2017-05-05:
This article is based on speculation.  A EULA doesn't say what is happening, it merely states what the company might be allowed to do.

Asus is the one you will have to ask, I don't have any technical information on how the Trend Micro features work.

closed 2017-05-05 by RMerl

diimdeep, 2017-05-05:
Out of the curiosity I started to look in code for clues..
Flags `TM_EULA` and `bwdpi_wh_enable` enable features that is mentioned in article.  
`check_bwdpi_nvram_setting` is used to check whether to use these features.
other functions is used when it is enabled.

```
TM_EULA
bwdpi_wh_enable
	check_bwdpi_nvram_setting
		ctf_fa_mode
		start_iQos
		start_dpi_engine_service # no source code  libbwdpi.so	
		start_firewall
		run_engine
			run_dpi_engine_service # no source code libbwdpi.so	
		add_iQosRules
			add_bandwidth_limiter_rules
			add_qos_rules
		start_wrs # no source code  libbwdpi.so	
		start_dc # no source code  libbwdpi.so	
```

libbwdpi.so

joehillen, 2017-05-05:
Is there anything that can be done to the firmware (proprietary or not) to prevent the information being leaked to a 3rd party? It seems irresponsible to shrug this off.

RMerl, 2017-05-05:
If you don't like the EULA terms, simply don't use any of the Trend Micro services - it's already fully in the end-user's control.  My point is, the original article is based on pure speculations, with zero investigation being made to determine what services requires what information to be sent to Trend Micro.  My personal gut feeling, this only happens when using the malicious website detection (WRS), as URL validation must be done server-side.  There's no technical reason (and no proof to that either) that Adaptive QoS would do the same - the router is already downloading signature files that are used locally by Adaptive QoS.

Panicking without having actual facts is what is actually irresponsible.  A EULA is not a fact, it's a just contract/agreement.  And speculating would be a waste of everyone's time - go to the source if you need answers.  I don't have any to give you.

joehillen, 2017-05-05:
Thank you. That's more the kind of explanation I was hoping for.

I did look at the source code and from my limited understanding it looks like it behaves like the EULA implies, i.e. with checks that any of the special services are enabled and then calling to binary blobs if they are. I have to defer to your expertise for more.

I particularly like QoS and bandwidth monitoring, and I want to use a firmware that has them, but I don't want my web history or other behaviour sent to a third party. I don't feel like that is a trade-off we should have to make.

If there was proof that the accusations are true, would you take steps to purge that behaviour from the firmware?

RMerl, 2017-05-06:
```If there was proof that the accusations are true, would you take steps to purge that behaviour from the firmware?```

That's a big "if".  But even if it were true, (which I doubt), there's nothing I can do about it as the code is closed source.  So no, I wouldn't be able to modify those component's behaviour, and I wouldn't remove the whole feature set either just because they need to leverage the cloud and people are uncomfortable with the idea.  That would go against the guidelines by which I manage this project, which is to remain close to the original firmware, and retain all of its original features.  I didn't remove AiCloud when security concerns arose about it, I didn't remove the WTFast feature for which a lot of users expressed skepticism, and so I wouldn't remove the Trend Micro-related features either.