Asuswrt-Merlin/asuswrt-merlin.legacy #1357: Disable the admin account for the openvnp service.

neoatomic, 2017-05-17:
Would it be possible to implement a feature to disable the admin/main account for the openvpn server?
I'm using a separate account that has a long complex password to harden the openvpn service. And therefor want to disable the admin account.

Also the new feature to assign ip's to a forwarded port looks promising.
But i have about 10 peers that need access on a specific forwarded port, not sure if it could be extended a little more. (and if possible Ddns entry's.)

RMerl, 2017-05-18:
Preventing admin access would be tricky, since OpenVPN handles authentication using PAM, not through a separate dedicated user database.

The port forward change was done by Asus.  I can't really extend on it, as it would break backward compatibility with them (for instance if I made it accept a CIDR).

neoatomic, 2017-05-18:
As the admin account was populated automatically in the openvpn service i already suspected that. ;(
But in case you see a possibility, it would be a nice addition. ;)

closed 2017-05-18 by neoatomic