Asuswrt-Merlin/asuswrt-merlin.legacy #1359: "Force as Master Browser" setting for samba shares not functioning on RT-AC3200

peterdimou, 2017-05-20:
Prior to the 380.66 and 380.66_2 update, I was able to access both of my USB attached drives through Samba using their drive labels when the "Force as Master Browser" setting was enabled. Now this no longer works and instead every top-level folder is listed under my router when attaching network storage in Windows. I have tried rebooting the router, enabling and disabling the setting, and removing the dot files on both drives. No combination seems to work and I think this might be a bug that was introduced as part of the new firmware.

RMerl, 2017-05-27:
This looks normal to me.  I have always had all the sub-folders listed here since day one, permissions set on the webui determines who can access which folder.

This is also completely unrelated to Browser Master, which merely determines which specific device contains the list of all devices with file sharing enabled - it does not handle the list of shares within the devices.  Share enumerating is done by each device itself when accessed.

peterdimou, 2017-05-27:
I won't go through the process of downgrading my router to confirm this issue, but I will say that in the past I was able to access the devices through their device label via Samba share. For example: if I named my router with the hostname "RT-AC3200" and it had two devices connected via USB labeled "2TB" and "4TB", I would be able to access both drives as a folder under the "RT-AC3200" device in Windows. With this firmware update, I can now only access the root-level folders on each drive and not the drives themselves. I believe in the "Browser Master" setting was the one that toggled this, but I could be mistaken. Anyway I can understand if this is low priority, but it is a feature I cared about and was convenient to me.

RMerl, 2017-05-30:
Root-level sharing always only worked in Guest mode.  In Authenticated mode, Samba only shares the folders at the root level.  You were probably using Guest mode in the past.

peterdimou, 2017-05-31:
You might be right. Unfortunately I won't be able to test that theory myself since I've recently moved back to the stock Asus firmware for unrelated reasons. Thank you for your time and attention to this issue and for all your hard work on the firmware.

closed 2017-05-31 by peterdimou