Asuswrt-Merlin/asuswrt-merlin.legacy #1360: No default route on IPv6 native

igloocube, 2017-05-20:
Device: RT-AC68U
Firmware: 380.66_2

Configured as:

[image]

IPv6 has no default route though and does not function.

If I SSH the router and do the following it works until the connection drops next:

ip -6 route add 0::/0 dev ppp0
ip -6 route add default dev ppp0 metric 1

Others reporting it here:
https://www.snbforums.com/threads/bug-ipv6-dhcp-pd-does-not-work-with-64-prefixes-on-asuswrt.30870/page-2#post-324516

igloocube, 2017-05-28:
Any more info you need or anything you want me to try?

RMerl, 2017-05-29:
I can't test any of this IPv6-related code, so any fix will have to come from Asus.

igloocube, 2017-06-04:
Is there a way I can log this to Asus?

I have found this as a longer term fix so it comes back on reconnections etc. Would nice to see it fixed in the base code though :)

Create wan-start user script in /jffs/scripts/. Add:
```
#!/bin/sh
ip -6 route add 0::/0 dev ppp0
ip -6 route add default dev ppp0 metric 1
```

ankhazam, 2017-06-04:
I believe for the moment adding the script contents as default actions post wan-start would be enough ;)

nimish, 2017-07-12:
Fixes my IPv6 issue. Would be nice to have a little toggle to default to 0::/0 as a default route if none is found.

RMerl, 2017-08-06:
Have you tried experimenting with the DHCP routing modes under LAN -> IPTV?

ankhazam, 2017-08-06:
Hm.... the found wan-start script's solution somewhat made us forget about the issue ;)

nimish, 2017-08-10:
@RMerl doesn't help, unfortunately

Zzyzx-Wolfe, 2017-08-22:
You can try contacting ASUS to see if they can fix the issue in a later firmware release here: https://vip.asus.com/VIP2/Services/QuestionForm?lang=en-us

nimish, 2017-08-29:
Looks like this might be an Asus issue, good news is they appear to have fixed it in 382. Short answer is that hardware CTF messes with IPv6. https://vip.asus.com/forum/view.aspx?id=20170823000125732&board_id=11&model=RT-AC5300&page=1&SLanguage=en-us

ankhazam, 2017-08-29:
well with CTF disabled I still need the wan-start script to have proper routing for IPV6 on my AC87U