Asuswrt-Merlin/asuswrt-merlin.legacy #1361: Network map not showing items within entire subnet

thomasvnl, 2017-05-25:
Hi,

I am using Asuswrt Merlin on two Asus RT-ACT66U models and I've discovered something funny. I just reconfigured my network to have a subnet of 255.255.252.0, which gives me a theoretical DHCP range of 192.168.0.1 - 192.168.3.254. In this case I have reserved the first 50 addresses, so 192.168.0.1 - 192.168.0.50, for static IP addresses, if needed now and in the future. '

Configuring these options all went fine and it works really well, however now comes the interesting thing I found. The network maps on both Asus routers only show items that are within the 192.168.0.0 - 192.168.0.255 range. Items outside this range aren't shown at all, even tough they are there. With an nmap scan I am able to find devices that are outside this /24 range.

Included at the bottom of this message I have included images of what the nmap scan found and what the Asuswert-merlin dashboard shows me. For the record, I am using version 380.65_4. Looking at the changelog and at the open issue list I couldn't find anything related between this version and the latest version that covered this so I felt confident enough to post a new issue here.

[image]

[image]

RMerl, 2017-05-25:
This is a technical limitation of the networkmap service, Asus hardcoded it for a /24.  Changing this would require major changes to the existing networkmap code, which would be problematic.

I know Asus has a new networkmap service in the next major firmware upgrade from them, hopefully they will have removed that limitation.

closed 2017-05-25 by RMerl

thomasvnl, 2017-05-25:
Ah okay, well that's unfortunate. Thanks for clearing that up. Looking forward to that next major upgrade. Any idea when that would be released (roughly)?

RMerl, 2017-05-27:
No idea, I don't know either if that new code will be compatible with past models, and if so with which model.