Asuswrt-Merlin/asuswrt-merlin.legacy #1364: RT-AC87U Web interface Changed from HTTP to HTTPS web interface not responding hanging

girotel, 2017-05-26:
Hi,

Noticed following problem.

Changed Web interface Change from HTTP to HTTPS web interface can not be reached anymore.

Web interface not responding accessible anymore

ROUTER: RT-AC87U 
Firmware: 380.66_2 (16_May-2017)
Browser: 52.1.2 Firefox Extended Support Release ESR
OS: Windows 7

Steps taken:
Administration - System
Web interface
Authentication Method
Change from HTTP to HTTPS web interface not responding

https://192.168.1.1:8443 not responding
Tried with different browsers like Chrome same problem.
Browser hangs

Factory default only solution to get web interface back.

Hope this could be resolved.

Thank you

TheUntouchable, 2017-05-26:
Have the same problem, but didn't connect it to the https till now. Funny thing: On different computers, there is no problem with the same version of firefox and plugins. Allready cleared cache and removed certifactes.

girotel, 2017-05-26:
Hi  TheUntouchable,

Question when you change to https you can reach the web interface with different browsers?

TheUntouchable, 2017-05-26:
Will test that this night!

RMerl, 2017-05-27:
Works properly for me.

TheUntouchable, 2017-05-27:
Okay, just tested:
Changed the web access to both, tried it without https: Works without any problems!
Tried it with https and port 8443: Firefox takes minutes to load the warning page for the certificate and after I allowed that my firefox freezes.

TheUntouchable, 2017-05-27:
Definitly a firefox problem.
1. Launched firefox with disabled plugins --> Doesn't work
2. Completely renewed my firefox profile --> Works without problems

Funnily it worked the first times since I got this router last week and suddenly stopped.

kokspflanze, 2017-05-27:
@TheUntouchable did you try with default https port 443? maybe firefox has just a problem with the 8443 port on https.

i tested it with firefox and chomium with port 443

TheUntouchable, 2017-05-27:
Sadly it does not change anything to use the port 443 for the webgui. It takes ages for the certificate warning page and so on.. :(

girotel, 2017-05-29:
Hi.

Thank you for testing.

Will try firefox in private window and see what happens.

nuintun, 2017-06-11:
@TheUntouchable HTTPS realy have some bugs, see:

[image]
[image]

When page refresh fast request will stalled!

RMerl, 2017-06-26:
I'm aware of stalls randomly occurirng over https (and they seem more frequent with Chrome).  However I have no idea what is causing them, a study of the SSL code in the firmware didn't reveal anything.  The SSL support in httpd (as implemented originally by the Tomato devs) looks a bit "fishy" at times, and might possibly require a major rewrite to potentially address those random issue - something that's beyond the scope of this project.

ptobis, 2017-09-19:
Hi,

I can confirm that the same issue occurs with firmware version 380.68_2 on Chrome 61 (win32).

xabix, 2017-09-24:
Hi all,

I have the same issue. Like gmail in https work but not yahoo mail and other sites in https. Not sure what to do to find out the issue but it's not a routing or DNS issue.

Merci

RMerl, 2017-09-24:
@xabix, your issue has nothing to do with the router firmware (or this issue entry).

MichaelEFlip, 2017-12-09:
Same problem here with RT-AC56U 380.68-4.
HTTPS-Webinterface is not accessible via different browsers. TLS handshake takes forever.
Is there a possibility to switch from HTTPS to HTTP via console to avoid factory reset?

xcjs, 2017-12-18:
I'm seeing the same issue for the RT-N66U with the web interface on port 8443 and HTTPS enabled. Is there a separate issue tracking that model?