Asuswrt-Merlin/asuswrt-merlin.legacy #1370: Guest wifi users have access to subnets connected through OpenVPN

MichaelEFlip, 2017-06-04:
Users of the guest wifi have access to subnets which are connected through OpenVPN.
Connections to the local LAN are prohibited, access to VPN not.

Syst3mSh0ck, 2017-09-27:
What router and firmware version. How did you determine this? Thanks.

MichaelEFlip, 2017-10-05:
Route is Asus RT-AC56U with currently 380.67-0.
OpenVPN connection is net2net. Problem occurs on the client side.
I can access hosts on the remote subnet (e.g. SSH).
Routes a configured manually (pushed by the server) - no NAT on tunnel.

haa-zee, 2017-12-28:
IMHO it is a serious bug in the firmware, because it handles the router's own subnet as local network, every other subnets as if they were on the internet. Because of this behavior the router denies access to other subnets for every clients which are blocked from internet, but it will allow these clients to access other subnets on the LAN. (https://github.com/RMerl/asuswrt-merlin/issues/1550)