Asuswrt-Merlin/asuswrt-merlin.legacy #1372: [AC66U] 380.67_alpha2:  The Country Code is not exist! Please enter Country Code.

mike-pt, 2017-06-05:
Model: AC66U
Firmware Version: 380.67_alpha2-g1bd1e71

After upgrading to 380.67_alpha2 I get this popup message when I go the Wirelesss Settings.

"The Country Code is not exist! Please enter Country Code."

And 5Ghtz wireless is not working (it was the only one I had enabled)

Some lines from the log that seem interesting
```
Jun  5 14:50:39 rc_service: httpd 346:notify_rc set_wltxpower;restart_wireless
Jun  5 14:50:40 kernel: Interface eth2 doesn't exist
Jun  5 14:50:40 kernel: Hooks already unregistered
Jun  5 14:50:40 kernel: Hooks already unregistered
Jun  5 14:50:41 kernel: wl_module_init: passivemode set to 0x0
Jun  5 14:50:41 kernel: eth1: Broadcom BCM4331 802.11 Wireless Controller 6.30.163.2002 (r382208)
Jun  5 14:53:38 dnsmasq-dhcp[607]: DHCPREQUEST(br0) 1.1.1.1.1 xx:xx:xx:xx:xx
Jun  5 14:53:38 dnsmasq-dhcp[607]: DHCPACK(br0) 1.1.1.1.1 xx:xx:xx:xx:xx:xx
Jun  5 14:54:17 rc_service: httpd 346:notify_rc restart_wireless
Jun  5 14:54:19 kernel: Interface eth2 doesn't exist
Jun  5 14:54:19 kernel: Hooks already unregistered
Jun  5 14:54:19 kernel: Hooks already unregistered
Jun  5 14:54:19 kernel: wl_module_init: passivemode set to 0x0
```

Resetting the settings doesn't seem to help, so I'm reverting back to the last stable.

RMerl, 2017-06-11:
Can't reproduce here.

This error message is typically caused by corrupted nvram settings. You might need to do a factory default reset and manually reconfigure your router.

The nvram settings you tried to change cannot be edited by the user, as they are hardcoded by the bootloader.

mike-pt, 2017-06-11:
I did twice and I keep seeing this, I also re-downloaded the firmware in case it was a bad download, but the same issue remains.

mike-pt, 2017-06-12:
I flashed `380.66_4` just now, and even before resetting I don't see the error and I can connect see the SSID. 

Since you can't repro I don't see what more can be done, I'll wait for a new build for .67 in any case and then try again.

Feel free to close If u don't think its worth debugging further.

mike-pt, 2017-06-17:
Just a note on the last comment, this was `380.67_alpha2-g925f2ea` and the main difference is that the popup is now gone (even just after a regular update, no settings reset), so definitely  something changed, from the previous build... (wireless is still not working)

I haven't reset yet, but will do to verify if the wireless is back, but just wanted to leave the note in case you find it interesting (maybe something did change between those builds)

RMerl, 2017-06-24:
This was probably caused by the missing AC66U binary blobs for newer GPL.  A kludge has been implemented since then to work around enum changes made by Asus in those newer GPLs.

closed 2017-06-24 by RMerl

mike-pt, 2017-06-25:
Just a quick note to confirm it's indeed fixed (tested with and w/o reset )