Asuswrt-Merlin/asuswrt-merlin.legacy #1380: Firmware: 380.66_4 / Router: RT-AC66U - OpenVPN Client - DNS queries blocked in VPN tunnel

singhandsingh, 2017-06-11:
Firmware: 380.66_4 / Router: RT-AC66U - OpenVPN Client GUI Settings Bug

I upgraded to 380.66_4, did a firmware reset and reconfigured my router manually with the same settings. 

Everything was perfect with the exception of the OpenVPN client configuration to StrongVPN. I managed to redirect all traffic through the tunnel after some initial issues. I confirmed this by running tracert to remote IP addresses. However, hostnames queries were failing and I realised that i was not able to perform nslookup to any DNS server, including the ones provided by StrongVPN. It seems that UDP/53 is blocked altogether.

I have set the following setting - 

>Accept DNS Configuration -> Exclusive
>Redirect Internet Traffic -> Policy Rules

Is this a bug ?

iMocW, 2017-06-12:
My OpenVPN does well.(Static key Auth)
Try to check your configuration.
OpenVPN aims to create a tunnel. Check iptables for NAT.

RMerl, 2017-06-26:
It's probably an issue caused by StrongVPN's servers, which don't push a proper default route.

See the following [discussion](https://www.snbforums.com/threads/split-tunneling-question-on-asuswrt-merlin.39824/) for a workaround.

RMerl, 2017-06-29:
Your configuration is missing a "redirect-gateway def1" statement, as it didn't get properly imported when you imported the ovpn file.  Add it to the Custom settings section, then restart your client.

singhandsingh, 2017-08-14:
Hi, i have bought a new router since my initial post. Was originally using AC66U. My new one is AC87U. I configured openvpn the same way as i did previously.

I am facing the same issue. I performed various trial and errors and i think the issue is real. Here are my findings.

Here were my initial configuration ..
a) Accept DNS Configuration -> Exclusive
b) Redirect Internet Traffic -> Policy Rules
c) Block routed clients if tunnel goes down: Yes
d) [In Custom Config]: dhcp-option DNS 8.8.4.4

1) When i change (a) to "Disabled", my DNS queries start to work. However, i fail the DNS leak test (dnsleaktest.com).

2) At this moment, i change it back to "Exclusive", and my DNS queries work for a short while and at this stage, i am able to pass the DNS leak test (with absolutely no trace to my country of residence). I am able to run nslookup with no timeouts at all, which is good news... After a short while, queries start getting blocked again...

3) While (2) was working, i decided to test (c) by switching of my fiber internet termination point. My asus router is still switched on.. and i am continuously pinging... 8.8.8.8. The results are as follows...

#### WHEN STILL CONNECTED TO THE TUNNEL, THE RTT IS HIGH. EXPECTED RESULT ####

Reply from 8.8.8.8: bytes=32 time=471ms TTL=58
Reply from 8.8.8.8: bytes=32 time=598ms TTL=58
Reply from 8.8.8.8: bytes=32 time=341ms TTL=58
......

#### WHEN FIBER TERMINATION BOX IS SWITCHED OFF. EXPECTED RESULT ####

Request timed out.
Request timed out.
Request timed out.
......

#### WHEN FIBER TERMINATION BOX COMES ONLINE. FAIL !!! RTT IS LOW AND THAT MEANS TRAFFIC LEAKS OUTSIDE TUNNEL ####

Reply from 8.8.8.8: bytes=32 time=7ms TTL=48
Reply from 8.8.8.8: bytes=32 time=6ms TTL=48
Reply from 8.8.8.8: bytes=32 time=4ms TTL=48
....

#### SUBSEQUENTLY. AFTER A FEW SECONDS... TRAFFIC NEVER FLOWS THROUGH THE TUNNEL UNTIL I MANUALLY START/STOP THE SERVICE ####

Reply from 192.168.1.1: Destination net unreachable.
Reply from 192.168.1.1: Destination net unreachable.
Request timed out.
Reply from 192.168.1.1: Destination net unreachable.
Request timed out.
Request timed out.
Request timed out.
...
 #####

So i see two issues here, it seems that DNS queries are prevented from going through the tunnel. Secondly, when the WAN comes up, traffic doesn't flow through the tunnel again.

Any thoughts..

singhandsingh, 2017-08-14:
Hi, i have bought a new router since my initial post. Was originally using AC66U. My new one is AC87U. I configured openvpn the same way as i did previously.

I am facing the same issue. I performed various trial and errors and i think the issue is real. Here are my findings.

Here were my initial configuration ..
a) Accept DNS Configuration -> Exclusive
b) Redirect Internet Traffic -> Policy Rules
c) [In Custom Config]: dhcp-option DNS 8.8.4.4

1) When i change (a) to "Disabled", my DNS queries start to work. However, i fail the DNS leak test (dnsleaktest.com).

2) At this moment, i change it back to "Exclusive", and my DNS queries work for a short while and at this stage, i am able to pass the DNS leak test (with absolutely no trace to my country of residence). I am able to run nslookup with no timeouts at all, which is good news... After a short while, queries start getting blocked again...

#### UPDATE - NEW RESULTS POPPED UP WHILE I WAS TYPING THIS ####

I manually added the following policy rules and my DNS qeuries now work fine ! Why didn't i think of this earlier !!

#Desc	Source		Dest		Iface#	
DNS1	0.0.0.0 	8.8.8.8		VPN

--

So this case is resolved with the above workaround.

I will post the NON DNS issue that i discovered in a separate thread.

singhandsingh, 2017-08-14:
UPDATE - While the workaround works generally fine, intermittent issues of DNS queries failing. It resolves itself when i repeatedly run the DNS queries via nslookup (or refresh my browser).

closed 2017-09-09 by RMerl