Asuswrt-Merlin/asuswrt-merlin.legacy #1384: Accessing WebUI using devicename/shortname denies access

seant-me, 2017-06-20:
Router: RT-AC68U
Firmware: 380.66_4

Relevant settings:
1. LAN - Device Name = asus
2. Administration - System - Authentication Method = HTTPS

Expected behavior:
I should be able to login and make setting changes using https://<devicename> OR https://asus

Actual behavior:
Sometimes I can login and sometimes I can't. Trying to make a change brings up the login dialog and doesn't make the change.

RMerl, 2017-06-21:
Works for me.  Try a different browser.

seant-me, 2017-06-23:
only works with the FQDN. https://asus.example.com works fine. https://asus will get me the login dialog but won't let me login. 

Tried chrome and safari browsers.

RMerl, 2017-06-26:
Still unable to reproduce it here.  You can give the 380.67 beta builds a try, Asus changed portions of the referrer validation code in it.

seant-me, 2017-06-29:
I just tried out 380.67 beta1 and same behavior. Is there a log I can send you?

RMerl, 2017-06-29:
No, as the web server doesn't log anything.

All I can suggest at this point is to stick to using a FQDN, as I'm unable to debug this, and Asus is also starting move portions of the referrer handling into closed source components, limiting even more ability to look at that code.

seant-me, 2017-06-29:
fair enough. thanks for looking.

closed 2017-06-29 by seant-me