Asuswrt-Merlin/asuswrt-merlin.legacy #1385: HTTPS redirect for webUI

seant-me, 2017-06-20:
Router: RT-AC68U
Firmware: 380.66_4

Relevant settings:
1. Administration - System - Authentication Method = HTTPS

Expected behavior
if i browse to http://<webui> but have HTTPS instead of HTTP enabled then the webui should redirect to the proper address https://<webui>

Actual behavior:
Chrome: This site can’t be reached. refused to connect.

RMerl, 2017-06-21:
Redirections are a bad idea for many different reasons.  For instance, if someone disables http access, then he expects the router to completely stop listening on port 80.  Otherwise, you are compromising some of the security benefits.  Also, automatically redirecting to the https port is the same as basically advertising which port to attack to any attack hitting the common port 80.

closed 2017-06-21 by RMerl

seant-me, 2017-06-29:
I could agree that if you absolutely didn't want anything to listen on port 80 then that would be the desired behavior -- especially if there was some potential port conflicts?? However, I'm also not allowed to access the webui over https on port 80. I get an error if I select HTTPS and port 80. 

It's also not any more or less secure. Changing ports and thinking you're anymore secure has been proven false (See security through obscurity). A simple port scan will identify many externally facing services like http(s). What's more secure is operating over https, timeout after X failed login attempts, or blacklisting offending IPs, etc.

unsynaps, 2017-06-29:
> However, I'm also not allowed to access the webui over https on port 80. I get an error if I select HTTPS and port 80.

Because Asus routers do not use port 80 for the management interface.