Asuswrt-Merlin/asuswrt-merlin.legacy #1386: Uploaded images/icons don't get stored in gallery

seant-me, 2017-06-20:
Router: RT-AC68U
Firmware: 380.66_4

Expected Behavior:
Adding icons to DHCP via upload works great. However, they don't get added to the gallery even though they are stored on the router for reuse. 

Actual Behavior:
Add icon to a dhcp reservation. adding the same icon to another reservation requires reuploading the same image even though it's already stored on the router.

rkkoszewski, 2017-09-30:
What do you mean with gallery? Is it the list of icons that come shipped with the router? 
If you mean that list then bad news, that list is actually static (hard coded in pure HTML and CSS). It is not dynamically generated and therefore won't show any uploaded images. I made once a pull request adding a few more default icons there, but that as far as it goes.

RMerl, 2017-10-07:
AFAIK this is simply how Asus designed that feature.

closed 2017-10-07 by RMerl