Asuswrt-Merlin/asuswrt-merlin.legacy #1389: FW:380.66_6 RT-AC87U Kernel panic after upgrade

imavroukakis, 2017-06-23:
After upgrading from 380.66_4 to _6, once the server reboots the logs record a kernel panic

```
Aug  1 03:00:14 kernel: CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c53c7f
Aug  1 03:00:14 kernel: CPU: VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
Aug  1 03:00:14 kernel: Machine: Northstar Prototype
Aug  1 03:00:14 kernel: Ignoring unrecognised tag 0x00000000
Aug  1 03:00:14 kernel: Memory policy: ECC disabled, Data cache writealloc
Aug  1 03:00:14 kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 60416
Aug  1 03:00:14 kernel: External imprecise Data abort at addr=0x0, fsr=0x1c06 ignored.
Aug  1 03:00:14 kernel: Mount-cache hash table entries: 512
Aug  1 03:00:14 kernel: CPU1: Booted secondary processor
Aug  1 03:00:14 kernel: Found a AMD NAND flash:
Aug  1 03:00:14 kernel: Total size:  128MB
Aug  1 03:00:14 kernel: Block size:  128KB
Aug  1 03:00:14 kernel: Page Size:   2048B
Aug  1 03:00:14 kernel: OOB Size:    64B
Aug  1 03:00:14 kernel: Sector size: 512B
Aug  1 03:00:14 kernel: Spare size:  16B
Aug  1 03:00:14 kernel: ECC level:   8 (8-bit)
Aug  1 03:00:14 kernel: Device ID: 0x 1 0xf1 0x80 0x1d 0x 1 0xf1
Aug  1 03:00:14 kernel: bio: create slab <bio-0> at 0
Aug  1 03:00:14 kernel: PCI: no core
Aug  1 03:00:14 kernel: PCI: no core
Aug  1 03:00:14 kernel: PCI: Fixing up bus 0
Aug  1 03:00:14 kernel: PCI: Fixing up bus 0
Aug  1 03:00:14 kernel: PCI: Fixing up bus 1
Aug  1 03:00:14 kernel: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
Aug  1 03:00:14 kernel: pflash: found no supported devices
Aug  1 03:00:14 kernel: bcmsflash: found no supported devices
Aug  1 03:00:14 kernel: Boot partition size = 524288(0x80000)
Aug  1 03:00:14 kernel: lookup_nflash_rootfs_offset: offset = 0x200000
Aug  1 03:00:14 kernel: Spare area=64 eccbytes 56, ecc bytes located at:
Aug  1 03:00:14 kernel:  2 3 4 5 6 7 8 9 10 11 12 13 14 15 18 19 20 21 22 23 24 25 26 27 28 29 30 31 34 35 36 37 38 39 40 41 42 43 44 45 46 47 50 51 52 53 54 55 56 57 58 59 60 61 62 63
Aug  1 03:00:14 kernel: Available 7 bytes at (off,len):
Aug  1 03:00:14 kernel: (1,1) (16,2) (32,2) (48,2) (0,0) (0,0) (0,0) (0,0) 
Aug  1 03:00:14 kernel: Options: NO_AUTOINCR,NO_READRDY,BBT_SCAN2NDPAGE,
Aug  1 03:00:14 kernel: VFS: Mounted root (squashfs filesystem) readonly on device 31:3.
Aug  1 03:00:14 kernel: et: module license 'Proprietary' taints kernel.
Aug  1 03:00:14 kernel: Disabling lock debugging due to kernel taint
Aug  1 03:00:14 kernel: et_module_init: passivemode set to 0x0
Aug  1 03:00:14 kernel: et_module_init: txworkq set to 0x0
Aug  1 03:00:14 kernel: et_module_init: et_txq_thresh set to 0xce4
Aug  1 03:00:14 kernel: =================
Aug  1 03:00:14 kernel: robo chk regs:
Aug  1 03:00:14 kernel: ===================
Aug  1 03:00:14 kernel: (0x00,0x5d)Port 5 States Override: 0xfb
Aug  1 03:00:14 kernel: (0x00,0x5f)Port 7 States Override: 0xfb
Aug  1 03:00:14 kernel: (0x00,0x0e)Port 8 States Override: 0x0a
Aug  1 03:00:14 kernel: (0x00,0x08)Port 8 IMP Control Register: 0x00
Aug  1 03:00:14 kernel: (0x02,0x00) Global Management Config: 0x00
Aug  1 03:00:14 kernel: (0x02,0x00) 802.1Q VLAN Control 5: 0x00
Aug  1 03:00:14 kernel: (0x02,0x03) BRCM HDR Control Register: 0x00
Aug  1 03:00:14 kernel: (0x00,0x0b) Switch Mode Register: 0x06
Aug  1 03:00:14 kernel: eth0: Broadcom BCM47XX 10/100/1000 Mbps Ethernet Controller 6.37.14.126 (r561982)
Aug  1 03:00:14 kernel: wl_module_init: passivemode set to 0x0
Aug  1 03:00:14 kernel: wl_module_init: igs set to 0x0
Aug  1 03:00:14 kernel: wl_module_init: txworkq set to 0x0
Aug  1 03:00:14 kernel: eth1: Broadcom BCM4360 802.11 Wireless Controller 6.37.14.126 (r561982)
Aug  1 03:00:14 kernel: _ Reboot message ... _______________________________________________________
Aug  1 03:00:14 kernel: <1>Unable to handle kernel NULL pointer derefere�U�at virtual address 00000000
Aug  1 03:00:14 kernel: <1>pgd = ceef8000
Aug  1 03:00:14 kernel: <1>[00000000] *pgd=9dd50031, *pte=00000000, *ppte=00000000
Aug  1 03:00:14 kernel: <0>Internal error: Oops: 17 [#1] PREEMPT SMP
Aug  1 03:00:14 kernel: <0>last sysfs file: /sys/module/bw_forward/parameters/dev_wan
Aug  1 03:00:14 kernel: <4>module:  wl	 bfbbf000	 3938054
Aug  1 03:00:14 kernel: <4>module:  ebt_mark	 bf419000	 814
Aug  1 03:00:14 kernel: <4>module:  ebtable_broute	 bf01d000	 873
Aug  1 03:00:14 kernel: <4>module:  ebtable_filter	 bf012000	 1061
Aug  1 03:00:14 kernel: <4>module:  ip6table_mangle	 bf7f8000	 934
[... 37 lines cut ...]
Aug  1 03:00:14 kernel: <4>[<c008f27c>] (sys_delete_module+0x16c/0x218) from [<c003dac0>] (ret_fast_syscall+0x0/0x30)
Aug  1 03:00:14 kernel: 002 0a000010 (e5924000) 
Aug  1 03:00:14 kernel: <4>---[ end trace 858ec7923de9b8c7 ]---
Aug  1 03:00:14 kernel: <0>Kernel panic - not syncing: Fatal exception
Aug  1 03:00:14 kernel: <4>[<c0043ff8>] (unwind_backtrace+0x0/0xf8) from [<c02cbbac>] (panic+0x7c/0x1a8)
Aug  1 03:00:14 kernel: <4>[<c02cbbac>] (panic+0x7c/0x1a8) from [<c00416f0>] (die+0x1a4/0x1dc)
Aug  1 03:00:14 kernel: <4>[<c00416f0>] (die+0x1a4/0x1dc) from [<c004515c>] (__do_kernel_fault+0x6c/0x8c)
Aug  1 03:00:14 kernel: <4>[<c004515c>] (__do_kernel_fault+0x6c/0x8c) from [<c00452cc>] (do_page_fault+0x150/0x1ec)
Aug  1 03:00:14 kernel: <4>[<c00452cc>] (do_page_fault+0x150/0x1ec) from [<c003d3a4>] (do_DataAbort+0x30/0x9c)
Aug  1 03:00:14 kernel: <4>[<c003d3a4>] (do_DataAbort+0x30/0x9c) from [<c0386e0c>] (__dabt_svc+0x4c/0x60)
Aug  1 03:00:14 kernel: <4>Exception stack(0xceebfeb8 to 0xceebff00)
Aug  1 03:00:14 kernel: <4>fea0:                                                       ca315340 9a000081
Aug  1 03:00:14 kernel: <4>fec0: 00000000 00000000 ca315328 ca315340 ca315360 d5eeb000 bee1aca8 ceebe000
Aug  1 03:00:14 kernel: <4>fee0: 00000000 00000000 00000000 ceebff00 bf7819bc bf781448 a0000013 ffffffff
Aug  1 03:00:14 kernel: <4>[<c0386e0c>] (__dabt_svc+0x4c/0x60) from [<bf781448>] (parse_ip6_header+0x1cb6c/0x1cf24 [IDP])
Aug  1 03:00:14 kernel: <4>[<bf781448>] (parse_ip6_header+0x1cb6c/0x1cf24 [IDP]) from [<bf7819bc>] (devid_sig_destroy+0x1bc/0x21c [IDP])
Aug  1 03:00:14 kernel: <4>[<bf7819bc>] (devid_sig_destroy+0x1bc/0x21c [IDP]) from [<bf7822f4>] (devid_exit+0x18/0x24 [IDP])
Aug  1 03:00:14 kernel: <4>[<bf7822f4>] (devid_exit+0x18/0x24 [IDP]) from [<bf744750>] (detector_remove+0x3c/0x4c [IDP])
Aug  1 03:00:14 kernel: <4>[<bf744750>] (detector_remove+0x3c/0x4c [IDP]) from [<bf73c804>] (nk_sys_exit+0xac/0xc8 [IDP])
Aug  1 03:00:14 kernel: <4>[<bf73c804>] (nk_sys_exit+0xac/0xc8 [IDP]) from [<c008f27c>] (sys_delete_module+0x16c/0x218)
Aug  1 03:00:14 kernel: <4>[<c008f27c>] (sys_delete_module+0x16c/0x218) from [<c003dac0>] (ret[<c003d334>] (do_IPI+0x114/0x154f9d1b00
Aug  1 03:00:14 kernel: <4>3f80: c03c3fc0 000000 ^Cc3fc0 c003ebe4 c003ebe8 600000134)
Aug  1 03:00:14 kernel: <0>Rebooting in 3 seconds..
```
The router stays on the previous firmware revision and is functional.

RMerl, 2017-06-24:
The panic occurred during the shutdown.  This is common, as you are trying to run the shutdown procedure off a filesystem that has just been rewritten by the new firmware.  This is a technical limitation caused by the fact that Asus routers don't use two separate filesystems for live and updated firmwares.

closed 2017-06-24 by RMerl

imavroukakis, 2017-06-25:
I see, thanks. Curious then, that the router still remains on the previous firmware version.

e2xo, 2017-10-22:
My question is this.

Aug  1 03:00:14 kernel: eth1: Broadcom BCM4360 802.11 Wireless Controller 6.37.14.126 (r561982)

If you whois 6.37.14.126 as an IP it's the United States Department of Defense in Arizona. Odd.

e2xo, 2017-10-22:
6.37.14.126
Fort Huachuca 
US Department of Defense Network
--

unsynaps, 2017-10-22:
Ahhh heh. I think thats a version number not an IP address.