Asuswrt-Merlin/asuswrt-merlin.legacy #1394: WAN SSL Web UI access is no longer working properly.

ruymbeke, 2017-06-25:
Hello,
With at least the latest RT-AC3100 380.66.6 and 380.67.alpha3 FW
the WAN SSL Web UI access is no longer working properly.
Even after clearing the cache, cookies and offline data and
allowing for the router SSL certificate exception Firefox just hang
and Chrome complains that the web page is not using SSL.
As an example if both authentication method are enabled then
https://x.x.x.x:8443 fails to connect when http://x.x.x.x:8080 works just fine.
The official Asus FW and the previous Merlin FW releases are ok.
Can you please confirm the problem and consider a fix ?
Thank you very much for this great development.
Best Regards.
Gilles van Ruymbeke

RMerl, 2017-06-26:
Works for me.  Can you confirm that you have both httpd processes running?

```
ps w | grep http
```

ruymbeke, 2017-06-26:
Hi Eric,
Yes both are up, cf below.
Which web browser are you using ?
Which router and FW version are you running ?
Thanks for your reply.
Best Regards,
Gilles

admin@RT-AC3100-xxxx:/tmp/home/root# ps w | grep http
  567 admin     6896 S    httpds -s -p 8443 -i br0
  568 admin     8496 S    httpd -i br0
 5548 admin     1436 R    grep http
admin@RT-AC3100-xxxx:/tmp/home/root#

ruymbeke, 2017-06-27:
Hello,
It looks like that we are both correct.
On a clean install with the default setup the Merlin FW WAN SSL is working ok.
But it looks like that enabling the jffs file system is the Merlin FW feature that breaks the WAN SSL.
I repeated the whole process below twice to double check this weird behavior.
Can you please check this out as well and confirm my findings ?
Best Regards,
Gilles van Ruymbeke

In my case once the WAN SSL is broken the only way I found to recover is to follow the steps below:
1) Restart Firefox and clear the last hour cache cookies and offline data.
2) Restore the default NVRAM values and re-flash the original ASUS FW via the router recovery mode.
    Power up the router while holding the reset button depressed (using a sharp pen) for about
    10 sec with the WAN interface disconnected until the power LED blinks at a medium speed.
    Then set your PC IP to 192.168.1.2 as an example and use the router emergency recovery page on
    http://192.168.1.1 and restore the default NVRAM values and load the latest original ASUS FW.
    I used the RT-AC3100_3.0.0.4_380_7743-g2cf84e9.trx file.
    When complete verify the router is back online by loading the web page at http://192.168.1.1
3) Restore the router factory defaults via the emergency repair option (not the web interface)
    Power up the router while holding the wps button depressed for about 20 sec until the power LED
    blinks at a fast rate then release the wps button and wait for the router to complete the boot.
    When complete verify the router is back online by loading the web page at http://192.168.1.1
5) Setup the router by providing the login credentials and the WiFi SSID / pwd.
4) Enable both authentication methods (http and https) and enable the web access from WAN.
    Verify that you can login using SSL from the WAN interface: https://x.x.x.x/8443
5) Upgrade the router with the latest Merlin (beta1) FW from the router web interface.
    I used the RT-AC3100_380.67_beta1.trx file.
6) Verify that you can login using SSL from the WAN interface: https://x.x.x.x/8443
    You may reboot to verify that it still works after a router reboot.

Now let's try to break the WAN SSL:
7) Enable the Format JFFS partition at next boot option and reboot the router.
    Enable the JFFS custom scripts and configs option and reboot the router
8) I added the script below but I think that WAN SSL was already broken without it.
    enable ssh and login via ssh to execute the following command to enable PXE boot.
    echo "dhcp-boot=grldr,server.name,x.x.x.x" > /jffs/configs/dnsmasq.conf.add
    reboot the router to activate the jffs file system and the newly added script.
9) Verify that you can no longer login using SSL from the WAN interface: https://x.x.x.x/8443
    even after restarting Firefox and clearing the last hour cache cookies and offline data.
    and that the only way to recover is to start again from the step 1).

Obviously I am very interested in getting the jffs file system and custom script capabilities of the
Merlin FW as long as I do not loose the WAN SSL capability. Again can you please confirm my findings ?
Thank you very much for your time. Best Regards, Gilles

TheUntouchable, 2017-06-27:
There is an easier way: Just clear your firefox profile and everything will work again.. ;)

ruymbeke, 2017-06-28:
Hello,
It is not a web client configuration, profile setup or caching issue.
There is something weird with the Merlin FW (re-flashing with Asus FW works ok) when accessing the
web pages of the router via the WAN interface (not LAN) and WAN IP on the SSL port 8443 (8080 is ok).
Both httpds and httpd servers are running and are binded to the bridge br0 of the router.
Interestingly and as expected the web interface works ok when connecting to the router
from the LAN interface and using either the LAN IP or WAN IP addresses on the SSL port 8443.
Even more interesting: trying to access the web page from the WAN interface and WAN IP on port 8443
(which is not working) is blocking other request made from the LAN interface (which will also fail)
until the previous request on the WAN interface is canceled.
So can you please try to replicate and confirm my tests: the SSL connection from the WAN interface
(not the LAN) to the router WAN IP address on the SSL port 8443 (8080 works ok) is not working.
Best Regards,
Gilles

ruymbeke, 2017-07-03:
Hello,
FYI the latest Merlin 380.67_beta2 FW still have the problem of broken SSL WAN access.
Even after a clean install: starting with an emergency restore of the latest Asus FW,
restoring the factory defaults settings, upgrading to beta2 and enabling both WAN accesses.
This time it looks like that SSL may also be broken from the LAN. Can you please confirm my findings ?
Please try a true WAN SSL access which is not the same as SSL accessing the WAN IP from the LAN network.
The WAN IP of my test router is on my local intranet to be sure that there is no ISP filtering (nested routers).
Thanks,
Gilles

RMerl, 2017-07-03:
I have no problem at all accessing my router over WAN through SSL.  If the issue appears after you enable custom scripts support, then you have either a custom script or a custom config file breaking your router's webui.

ruymbeke, 2017-07-03:
Hello,
Thanks for your reply.
I have no custom script and JFFS is not enabled. My router is a RT-AC3100.
I performed a clean install: router was reset to default configuration and re-flashed to the
latest Asus FW via the emergency recovery minimal web interface after clearing the NVRAM.
Then after entering the router password, WiFi SSID and encryption keys (which are requested
after resetting to the default settings) the only thing I enabled are the Authentication Method
set to both and Enable Web Access from WAN set to yes. With the Asus FW WAN SSL is working ok.
Then after upgrading to the Merlin beta 2 FW (RT-AC3100_380.67_beta2.trx ) WAN SSL is broken.
Reverting back to the Asus FW (RT-AC3100_3.0.0.4_380_7743-g2cf84e9.trx) WAN SSL is again ok.
Re-flashing to Merlin beta 2 SSL is broken again.
Did you try these same steps and on the same  RT-AC3100 router ?
Again thank you very much for your reply.
Best Regards,
Gilles

huatux, 2017-07-04:
@ruymbeke   My AC5300 has the same problem when i use FireFox...but i got chrome on my laptop,so i try to access it with chrome,it works : )  hope this help you

ruymbeke, 2017-07-05:
Hello,
Thanks for your reply but this is a Merlin FW issue with at least this RT-AC3100 router.
Google Chrome, MS IE, MS Edge and Firefox show all the same issue. When reverting back
to the original Asus FW the SSL WAN connection works ok with all these web browsers.
Best Regards,
Gilles

RMerl, 2017-08-06:
So far no one reported anything similar, and your output confirms that the httpd daemon does run, so it's not caused by a corrupted SSL certificate either (which was a potential issue in 380.67, fixed for 380.68).  I suggest going to the support forums for further troubleshooting, as it seems more likely to be a client-side issue than a code issue.

closed 2017-08-06 by RMerl

ruymbeke, 2017-11-16:
Hello,
With the latest 382.1 FW (Asus RT-AC3100) is is expected to use the same SSL certificate
on both the WAN and the LAN interfaces ? This could explain some of my problems since
I can see that the SSL certificate used on the WAN is referring to the LAN IP address.
I also noticed that Firefox just times out without even complaining about it where
Chrome eventually complains about it but let you continue by installing the wrong certificate.
Interestingly when that happen the Processor usage of the router goes up to 100% for some
period of time (monitored by a non SSL LAN connection) . So something does not seems right.
Please comment and advise. BTW, thank you very much for your great work !
Best Regards,
Gilles