Asuswrt-Merlin/asuswrt-merlin.legacy #1398: A66U, password not remembered correctly

Kapot, 2017-06-28:
Hi there,

Not really sure if it is an error but wanted to let you know.

On my AC66U/VPN/Open VPN Clients page there seems to be some errors with 'special' characters like % $ & etc. Whenever I want to start the VPN connection after I turned it off it tells me the authentication was wrong. When I click the box 'show password' the password isn't right; it suddenly is a lot longer with stuff like '&#-32&#-68&#'.  I need to retype my password every time before I can connect again.

Also, the 'Please wait, 44% Applying settings' popup goes above 100%, only on this page. It usually goes to 170+ % for me. Minor thing tho ;)

Thanks in advance!

RMerl, 2017-06-29:
First issue seem related to form encoding.  I'll see if I can something about it.

Second one is random, and I have no idea what triggers it.  I have only seen it happen twice myself over the months.

RMerl, 2017-07-08:
Can you provide me with an example password causing the issue?  I just tested with "t1$t2%t3" here, and the password is correctly stored and retrieved from nvram.

Kapot, 2017-07-08:
Sure. Something similar to: &76VreFNW!

RMerl, 2017-07-09:
The issue is probably specific to the "&76" part, which gets interpreted as a meta entity by the web browser.  I'll see if it's possible to escape such sequences from it.

commit a08b0a99 (2017-07-09): webui: properly escape OpenVPN user/passwords that contain &xx strings (close #1398)
closed 2017-07-09 by RMerl