Asuswrt-Merlin/asuswrt-merlin.legacy #1403: Weird DNS issue (resolved to 10.0.0.1)

boringuy, 2017-07-03:
I have had this problem since the beginning and just upgraded to the latest 380.66_6 and problem doesn't resolve. I only had this router (RT-AC86U) for about 6 months and immediately flash the merlin firmware, so I am not sure if this is a problem from the original problem as well.

I have always thought this is cause by my ISP going down but it's happening more and more frequent and finally decided to dig a little deeper.

When the problem happens, I found that I can ping www.yahoo.com and do nslookup from ssh into the router itself:
`admin@RT-AC68U-4120:/tmp/etc# nslookup www.yahoo.com
Server:    127.0.0.1
Address 1: 127.0.0.1 localhost.localdomain

Name:      www.yahoo.com
Address 1: 2001:4998:44:204::a7 ir1.fp.vip.ne1.yahoo.com
Address 2: 98.139.183.24 ir2.fp.vip.bf1.yahoo.com
Address 3: 98.139.180.149 ir1.fp.vip.bf1.yahoo.com
Address 4: 98.138.253.109 ir1.fp.vip.ne1.yahoo.com
Address 5: 98.138.252.30 ir2.fp.vip.ne1.yahoo.com`

However, if I don't from my Mac (or my rasberryPi) there result is this:
`Andys-MacBook-Pro:~ andy$ nslookup www.yahoo.com
Server:		8.8.8.8
Address:	8.8.8.8#53

Name:	www.yahoo.com
Address: 10.0.0.1`

`Andys-MacBook-Pro:~ andy$ nslookup
> server
Default server: 8.8.8.8
Address: 8.8.8.8#53
> server 192.168.0.1
Default server: 192.168.0.1
Address: 192.168.0.1#53
> www.yahoo.com
Server:		192.168.0.1
Address:	192.168.0.1#53

Name:	www.yahoo.com
Address: 10.0.0.1
> www.yahoo.com
Server:		192.168.0.1
Address:	192.168.0.1#53

Name:	www.yahoo.com
Address: 10.0.0.1`

It doesn't matter what I change my /etc/resolv.conf to (8.8.8.8 or 192.168.01), it always return 10.0.0.1 from the Mac while I can ping 8.8.8.8 or 98.139.183.24(yahoo) without problem. So, it's not a problem with the ISP. I have try using the ISP's DNS server or Google's DNS server from the router setting but doesn't change anything.

I have to "service wan_restart" to get it to work again. Then, after a while, it happens again. Sometimes, it will work for a whole day without problem. Sometimes, it happens several times within an hour. 

If there is anything I can try to troubleshoot this problem, please let me know. Thanks!

RMerl, 2017-07-03:
This is most likely a configuration issue.  Please post on the support forums at SNBForums for troubleshooting.

closed 2017-07-03 by RMerl

boringuy, 2017-07-15:
After some troubleshooting, turning off the "Enable WAN down browser redirect notice" completely fixed the problem. Definitely a bug in that feature.