Asuswrt-Merlin/asuswrt-merlin.legacy #1405: openvpn tunnel encryption not working ios

duecedriver, 2017-07-03:
trying to setup openvpn on merlin latest

when trying to enble TLS control channel security and setting encrypt channel... I can get my macs running tunnelblick to conntect to the router no problem using the exported config but my ipad and iphones won't

using the openvpn clients on both and they are advertised as being compatible with the latest openvpn feature set?

am I doing something wrong or is this a known issue on iOS?

currently I have the second openvpn server setup with the same settings but have TLS control channel security set to disable and that works fine on ios... so its pointing squarly at tunnel encryption...

thanks

also .. with a internet connection of 60mbps down 6 up... running a speed check on the router is showing 100% cpu with tunnel encryption so I guess I am being cpu bound... should I overclock the n66u processor or live with it...

thanks guys

RMerl, 2017-07-03:
Please use the support forums at SNBForums for support requests.

closed 2017-07-03 by RMerl

unsynaps, 2017-07-04:
The iOS client does not support encrypt channel. No idea why but that is on the OpenVPN team to get working. Sucks cause I have to run two OpenVPN servers. One with it off for my iOS devices and one with it on for everything else. Other than that all the settings are the same between the two.

As for the speed that is on the routers CPU. Its just not beefy enough. I am on Fios (150/150) and the most I get from work over the VPN is ~45/~45 on an AC87 which runs at 1GHz. Even overclocked to 1.2GHz speed only gets bumped to ~50/~50. Even shutting off compression does not yield much more speed. Been meaning to test it with encryption off to see what I get but (1) im lazy (2) whats the point in having a VPN with the encryption off even if it is faster? :P

duecedriver, 2017-07-04:
@unsynaps 
thanks for the reply.. merlin just closed all my querries and pointed me at the forums.. thanks for getting back to me