Asuswrt-Merlin/asuswrt-merlin.legacy #1408: SQM QoS?

eversins, 2017-07-03:
Any chance we can get a simple SQM (Smart Queue Management) QoS to prevent buffer bloat? Is hardware acceleration always disabled if using QoS?

Thanks.

RMerl, 2017-07-03:
What can be done is limited by the old kernel Broadcom uses (2.6.36).  fq_codel was already backported, however it's only available on models using 2.6.36, and also currently only with Traditional QoS.  The RT-N66U and RT-AC66U kernel is much too old.

RMerl, 2017-07-08:
Closing this, as limited SQM was already available for Traditional QoS and now Adaptive QoS - this is the best that can be done with the current kernel version.

closed 2017-07-08 by RMerl

ChriscomIT, 2017-07-09:
Great to see the QoS improvements for Adapative QoS. Nice work Eric!!