Asuswrt-Merlin/asuswrt-merlin.legacy #1413: syslog format invalid

mhofman, 2017-07-08:
AsusWRT seems to sometimes generate invalid syslog messages.
According to [section 4.1.3 of RFC3164](https://tools.ietf.org/html/rfc3164#section-4.1.3), the TAG/program field of the message should not contain any space.

I've encountered 2 occurrences so far of this happening:
- `hour monitor`
- `custom script`

This is a problem when sending logs remotely and ingesting them with syslog-ng. See https://github.com/balabit/syslog-ng/issues/1591 for more details.

Would it be possible to replace those spaces with a dash for example?

RMerl, 2017-07-08:
Shouldn't be a problem, as long as those instances (these two are) are within open-sourced code.

mhofman, 2017-07-11:
Thanks for the quick action.
I've encountered a few other occurrences since then:
- `WAN Connection`
- `ddns update`
- `dhcp client`
- `dhcp6 client`
- `Samba Server`
- `FTP Server`
- `WEBDAV Server`
- `NAT Tunnel`
- `disk monitor`

RMerl, 2017-07-11:
That's a lot of cases.  Rather than having to constantly chase after new breaking entries Asus might make, I'll see if logmessage() could be made to replace any space with an underscore character, which should automatically take care of it.

RMerl, 2017-07-11:
That will take care of any instance (including closed-source ones) where libshared's logmessge() is used.

There might still be some instances that directly call openlog() and which are closed-source - these would have to be fixed upstream by Asus (if any).

Keeping this issue open for now, in case more exception cases might be spotted.

RMerl, 2017-07-11:
BTW, note that RFC3164 is obsolete, and should be replaced with RFC5424.

mhofman, 2017-07-11:
That's a good approach! I had looked downstream at the `syslogd.c` file but didn't see a way to handle the tag there.
I guess that anything calling the `logger` utility with a specific tag containing a space will bypass this fix since it calls `openlog()` directly, but that's probably fine.

Regarding RFC5424, I agree but that's a change on the router side as well, right? I'm not too familiar with how logging works in linux, but at some point something must format the log message in a certain way, and it seems to be the BSD/RFC3164 way right now.

mhofman, 2017-07-11:
I was looking at the syslogd code again in Busybox, and I think I understand why syslog-ng is "overriding" the hostname when a space is in the tag: the log sent by syslogd in busybox to the remote is the raw log line read from `/dev/log` and not the formatted line. The raw line only contains integer priority, timestamp (optional) tag and message, without the hostname. So when a space appears in the tag, syslog-ng believes the first word is the hostname.

Maybe we should consider sending a formatted line to the remote in syslogd, that contains the hostname/IP and ensures the timestamp is present? Most of the code for this already exists in the `timestamp_and_log()` function.
My understanding is that syslogd is considered a relay between the local device and the remote server, and is thus responsible to format the message appropriately.

RMerl, 2017-07-12:
I'd rather not start messing with the busybox code, I already have more than enough code to deal with with the rest of the firmware, and I'm already not a fan of the amount of bb patching Asus and Broadcom have made over the years to bb - makes it a pain to keep it up-to-date.

themiron, 2017-07-12:
@mhofman right, and things can be improved if revert log smallification for sending entries out

mhofman, 2017-07-12:
@RMerl, searching through the code, I found 2 instances of `logger -t` being called directly with a space in the tag.
`logger -t "custom script"` in
- [others/vpnrouting.sh#L75](https://github.com/RMerl/asuswrt-merlin/blob/72cdf4d5a736ddc1fbff71ae1afcee5325bd5566/release/src/router/others/vpnrouting.sh#L75)
- [others/updown.sh#L110](https://github.com/RMerl/asuswrt-merlin/blob/2f48b2c6d501d7e0d3a7ccd252fdac2eff1e60d6/release/src/router/others/updown.sh#L110)

mhofman, 2017-07-13:
Between the explicit removal of some spaces and the logmessage function changes, I believe this issue can be closed.

As a temporary workaround, and extra layer of normalization, since the Synology Log Center I'm using to ingest the logs isn't the best at configuration flexibility, I also setup on my router a syslog-ng instance that works as a relay, normalizing the tag, and sending them using RFC5424.

I've posted the details of my config on the forum: https://www.snbforums.com/threads/configuring-syslog-ng-with-merlin-firmware.35095/page-2#post-335232

RMerl, 2017-07-13:
The current workaround should be good enough, until either syslog-ng or busybox can address the root cause.

commit d625d157 (2017-07-08): rc, shared: provide RFC3164 compliant TAG field for hour_monitor and custom_script logmessage() calls (closes #1413)
closed 2017-07-13 by RMerl