Asuswrt-Merlin/asuswrt-merlin.legacy #1414: Link to downloadmaster broken, renders URL with two ports as http://asus:8443:8081

michaelcoburn, 2017-07-10:
Hi, the link to downloadmaster under USB Application is broken, the URL successfully converts to non-SSL (http) but incorrectly continues to include port 8443, rendering the link as two ports.  As downloadmaster runs as a separate service on a different port, the link should be:
http://AsusRouter:8081
but on 380.66_6 it writes link as:
http://AsusRouter:8443:8081
See image below.
[image]
For some reason the host_name isn't getting split on the colon and throwing away the port 8443, perhaps here?
https://github.com/RMerl/asuswrt-merlin/blob/583fd39588955d4a8564a106cb64f5c34beb113e/release/src/router/www/APP_Installation.asp#L537-L541

RT-AC56U
Downloadmaster ver. 3.1.0.101

RMerl, 2017-07-10:
That code was changed by Asus with GPL 7627, it's possible that they broke things then.

RMerl, 2017-07-10:
With 380.66 I don't even get any link at all, while with 380.67 beta I do get a proper link, with only one port.  Can you retry please with 380.67 beta 3?

RMerl, 2017-08-06:
Assuming this was fixed with GPL 77xx as there has been no further feedback.

closed 2017-08-06 by RMerl