Asuswrt-Merlin/asuswrt-merlin.legacy #1415: NTP / DNS not working from Router

MarkusI72, 2017-07-12:
With 380.66_6 on my RT-AC68U I have issues with DNS and NTP.
With all clients, DNS works well but not on the router itself.
E.g. using network tools to ping Google or so fails resolving the hostname.
NTP does not sync at all (because it also has issues resolving the hostname of NTP server?)..

The symptom was that OpenVPN did not work anymore (certificate not valid yet) because the router server time was back somewhen in 2015...

hoerbi1000, 2017-07-15:
yes asus ddns also not working for me

MarkusI72, 2017-07-18:
Same with 380.67_0, DNS not working, NTP not working

MarkusI72, 2017-07-25:
Any update on this?

unsynaps, 2017-07-25:
Sounds like something specific to your router/ISP/configuration. Not sure about anyone else but I am having no problems.

MarkusI72, 2017-07-26:
So, your NTP works and your network tools (e.g. ping google) from within the router UI?
Clients do not show any issue.

MarkusI72, 2017-07-26:
Strange - for whatever reason - it somehow recovered from the issue...

closed 2017-07-26 by MarkusI72