Asuswrt-Merlin/asuswrt-merlin.legacy #1417: Regression: RT-N66U_380.67_beta4.trx cause endless router interface up/down loop

ache, 2017-07-14:
It is even hard to reset settings to default, because it up/down (or kernel panic?) too fast.
With setting resetted is stays few minutes, but then goes to the same loop in any case.
I was quick enough to upload RT-N66U_380.67_beta3b.trx when the router was in that semi-alive state.
It is clear regression because RT-N66U_380.67_beta3b.trx works normally.

ache, 2017-07-14:
I think initially that HE tunnel setting cause beta4 to endless restarts, because it happens when I apply them. But next time I reset settings and still have endless restarts despite of they are resetted. Only few more settings reset gives me window enough to upload beta3.

RMerl, 2017-07-15:
Can't reproduce here.  I flashed my RT-N66U with 380,.67 beta 4, did a factory default reset, configured Wifi and DHCP WAN.  After that I rebooted again, configured an HE tunnel, rebooted once more - still no issue at all.

ache, 2017-07-15:
I have Intel 82579V on my comp and internal driver utility which spy on driver status link speed shows that link goes from not connected to 1Gbps Full Dulpex about in ~3-4 seconds interval in the loop. How to make it stable with new beta at least to flash old beta3b still is not predictable well to me, I think it was an contingency. 
It never happens with old beta3b and newly written back beta3b, so it is not hardware problem at all. In Changelog I see you change GPL code from 7378 for N66U to just using binary blobs alone from 7378 for N66U, maybe it can play role.

RMerl, 2017-07-15:
The Ethernet driver hasn't changed in years.  I'd do a complete electrical reset of your router (remove its power for 30 secs, try turning it on while the power is unplugged to drain if of any residual charge, then plug everything back).  Also check the Ethernet connection - flapping is often caused by poor cable quality.

I haven't had a single other report of issues with the RT-N66U, so I doubt it's related to the firmware, especially as beta 4 brought no hardware change, it was mostly QOS-related change specific to newer models.

ache, 2017-07-15:
I doubt it is ethernet driver, it looks more like fast kernel panic/autorestart. The cable show perfect in Intel 82579V internal testing while the working window is available.
I'll try your recommendations in any case and report here. I have QOS turned off.

ache, 2017-07-15:
BTW, any recent firmware before beta3 works well too.

ache, 2017-07-15:
I do all steps you recommend, including replacing cables with new ones. The picture stays is exact the same...
In favor of my guess that it is not ethernet driver speaks log window, where I always see complete boot as always, no additional interface up/down change or additional corresponding jffs scripts running (for beta4). 
I also try to turn on QOS, but it does not help. Moreover, any settings reset (or reboot) from the web interface (even from CFE mini loader which comes sometimes instead of full interface) does not work, only hardware reset button on the router reset the settings, 
This time it was even harder for me to load beta3b back because even after fully loading of beta3b I still see myself on beta4 once. It was harder to find proper window this time too (as I already tell, settings reset was only temporary solution), so I decide to drop my efforts and stay on beta3b which works perfectly (this message was written on beta3b f,e).
Perhaps my info will be slightly useful for you.

ache, 2017-07-18:
BTW, it is side question. but which router  you recommend now for small home as most stable (without Qualcomm chips)? I don't want to sit on obsolete with time passed firmware forever.

alximix, 2017-07-19:
I have the same problem on 380.67. Just for the ac5300 router, wan interface drops after 15-20 minutes. Now I will try the beta. In previous releases of firmware no such problems were observed.

ache, 2017-08-14:
The same thing with RT-N66U_380.68_beta1-gbc6ee4f.trx too, nothing is changed. Very hard (see above) downgrade to RT-N66U_380.67_beta3.trx works perfect as always (when really happens). It looks like GPL 380_7743 code is incompatible with some very old RT-N66U routers like mine one and cause constant reboots by some kernel fault and NIC up/down visible is side effect of it because I see no NIC up/down in the log, only whole reboot.

siborg, 2017-08-22:
I have the same issue on RT-AC87U. For me the last working version is 380.65_4 . Anything after that and the interface goes into on/off mode. It connects then reconnects. Sometimes hangs the router altogether.

RMerl, 2017-09-09:
This is most likely unrelated to the firmware itself, as there are thousands of users with the same models (including myself, with one of the early RT-N66U revisions) without any issue.

closed 2017-09-09 by RMerl