Asuswrt-Merlin/asuswrt-merlin.legacy #1419: Loss of connectivity noticed on WiFi Clients during port scan

fdx90, 2017-07-16:
Doing a Shields up (all service ports) scan @ https://www.grc.com/x/ne.dll?bh0bkyd2 causes strange behaviour of AC88U running RT-AC88U380.67 Beta 4. WiFI such as WiFi clients loss of connectivity until scan ceases.

Able to reproduce.

HuckToohey, 2017-07-16:
How is that weird; isn't it what's supposed to happen when you run that particular portscanning script?

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

fdx90, 2017-07-16:
Hi,

Thanks for the quick reply.

Just so I'm clear. You seem to be saying this behavior is "by design" and you don't see any particular issue with how it's functioning. 

I'll confirm the experience with some more verbosity; What I'm describing as an issue is;

A port scan requested by a LAN client originating from the internet to the wan interface seems to impact LAN clients negatively, I've seen it cause wifi clients loose traffic flow, large lag spikes and disconnections from games and SaaS apps e.g. Drop box, Skype. Never seen this before testing other devices, and can't imagine how this would be a built in feature.

This is reproducible on my ac88u running the latest beta 4. The device also becomes sluggish and slow to respond on the web ui when attempting to access it.

To clarify, it's a "Scan yourself" web URL. (GRC shields up!) Not a script that I'm running locally.

On my end at least, this causes noticeable issues with lan clients, and router responsiveness noticed by the clients connected.

There are no changes to iptables, and no scripts that I use other than a killall -9 infosvr on service-start.

There is some component of that port scan that's impacting device performance, I'll happily test a non Merlin firmware if there's a version I can flash to and let you know if it continues that you'd recommend.

I'm not a network guru but can manage the basics well enough, let me know if you need anything from me to help test.

Cheers,

FDX

jackyaz, 2017-07-19:
Do you have DoS protection enabled in the firewall settings, I imagine it's limiting packet flow and the ShieldUp scan is eating all of the available flow.

Also, post here to get a faster answer: https://www.snbforums.com/forums/asuswrt-merlin.42/

RMerl, 2017-08-06:
You are probably DDoS'ing yourself with the scan, especially if you enable logging.

closed 2017-08-06 by RMerl