Asuswrt-Merlin/asuswrt-merlin.legacy #1420: User-generated certifacted stored in /jffs/ssl is ignored

mieszkomularczyk, 2017-07-17:
Hello,
I use 380.67 on RT-AC68U.
Use persistent certificate is set to YES
Generate a new certificate is set to NO
I replace existing cert.pem and key.pem with my own files and reboot the router.
After the reboot, my files are overwritten to the default Asus-Merlin self-generated certifcate. Before FW 67 I used this https://www.rickygao.com.au/blog/how-to-upload-ssl-certificate-to-asus-router/ and everything was fined.
It looks like /jffs/ssl certifacte storage feautre introduced recently does not work properly.

Trelor, 2017-07-19:
I am experiencing this same issue on an RT-AC66U, I even tried the older way of editing the /etc/cert.pem /etc/key.pem and it does not work.  It also looks all settings are ignored or there is something hardcoded some where.

Lloyd1978, 2017-07-20:
Here's how I got it to work...

1. Log in via SSH
2. Stop the web server...

`# service stop_httpd`

3. Edit the key, I had to manually delete the contents, and paste the new key in (paste = right click in putty) making sure the begining and end dashes of the certs are on new lines...

`# nano /jffs/ssl/key.pem`

4. Edit the cert, manually deleting the contents of the certificate before pasting your new one as with the key...

`# nano /jffs/ssl/cert.pem`

5. Finally...

`# reboot`

And it should be set!
It's not pretty, but it worked, I'm now using my own ECC256 certificate, generated on a Windows CA and converted to OpenSSL.

While I'm here, small feature suggestion regarding setting our own SSL certificates...perhaps we should have text boxes to input our certificates like the authorized_keys have.

Anyway, I hope that helps someone else.
Kind regards

Lloyd

mieszkomularczyk, 2017-07-20:
@Lloyd1978 of course I did it like this first time etc. 
What finally helped me was:
`nvram unset https_crt_file`
rebooting the router and then copying the stuff to /jffs/ssl.
@RMerl I think this is a bug. In case of certificate/key presence in /jffs/ssl, the router should ignore all certifacte related settings in NVRAM. What do you think?

Lloyd1978, 2017-07-20:
Oh good, glad to hear you got it working :)

Trelor, 2017-07-20:
Thank you this method worked for me too.

RMerl, 2017-07-20:
@mieszkomularczyk I'll have to double check how I'm handling things.  I believe I refer to the nvram values so any existing content could be exported first, there might be some corner case where it's not being properly flushed afterward.

RMerl, 2017-07-21:
I had a look at the code, and https_crt_file is no longer used anymore.  If something is causing your router to use the content of that nvram setting then it must be a custom script you've added, not something from the firmware itself.

mieszkomularczyk, 2017-07-21:
@RMerl, thank you. After I managed to get it working,, by'nvram unset https_crt_file'  I am unable to recreate the problem. Thus I am closing the ticket.

closed 2017-07-21 by mieszkomularczyk