Asuswrt-Merlin/asuswrt-merlin.legacy #1421: All DNS traffic redirected to 10.0.0.1 while WAN is down (including internal hostnames)

Peter200lx, 2017-07-17:
I have a similar issue to #275. What setting on the WAN page controls the iptables DNS redirect when WAN is down? I have the `DNS and WINS Server Setting` configured to `Enable Manual Assignment`, but when my WAN connection went down the iptables rule was added to redirect all UDP port 53 traffic to my router port 18018. If I manually delete the rule then DNS resolutions inside my network work as expected, but that's obviously a temporary fix. I'm running firmware 380.65 on a RT-AC66U router.

jackyaz, 2017-07-19:
Please post the problem here https://www.snbforums.com/forums/asuswrt-merlin.42/

Peter200lx, 2017-07-19:
While looking at the forums before creating a post I found:
https://www.snbforums.com/threads/weird-dns-issue-always-resolved-to-10-0-0-1.39962/
Which talks about a `"Redirect to error page"` setting. When investigating further I found it under:
`Administration` > `System` tab > `Web Interface` section > `Enable WAN down browser redirect notice`
It was enabled and I've toggled it to disabled. Closing this issue as I believe this is the setting I was looking for.

closed 2017-07-19 by Peter200lx