Asuswrt-Merlin/asuswrt-merlin.legacy #1425: Feature Suggestion: Set Custom SSL Certificate in GUI

Lloyd1978, 2017-07-20:
Hi,
You've probably already thought of it and/or working on it, but what about if we could set the httpd certificate in the GUI, like we do for the SSH authorized_keys file.
One text box for the key and one for the certificate+intermediate.
Anyway just a thought. I hope this is the right place for a feature suggestion.
Kind regards

Lloyd

mieszkomularczyk, 2017-07-20:
Great idea, I fully support it!

RMerl, 2017-07-20:
I decided not to because due to the way the webui is implemented, anything provided through the webui will need to:

1) Use the same amount of nvram space for temporary storage.  That means a few KBs wasted just for a certificate
2) Be small enough to fit within the internal buffers used by the web daemon.

A basic certificate might not be so bad, however anything that includes an intermediate or a CA would be too large to fit.  Therefore that's why at this time, the only way to provide one is by putting it there through SSH/SCP/etc...

mieszkomularczyk, 2017-07-21:
@RMerl One way to avoid the problems you described could be (maybe?) doing it in GUI in a specific way.
Instead of edit_box for copy&paste of text-based key (like for SSH), you could solve it similar to firmware upload. There would be upload button and user would select a local with key and cert and the gui would simply upload it so /jffs/ssl.
Currently it is annoying for typical windows user: I need to ssh to the router and copy&paste using console and cat command or do scp from command line.
What do you think?

RMerl, 2017-07-21:
Form upload would indeed be a way (this is how Asus has recently implemented it for the BRT-AC828 - I only saw that model's new GPL code yesterday).  This will require a fair amount of work however, so it will have to wait a bit.  I might need to move that configuration to a different page however (for some weird reason, Asus is locating it on the DDNS page...)

RMerl, 2017-08-06:
Implemented with bc0b23fc36527676fca698f71234fb8d48c48fd4.

closed 2017-08-06 by RMerl