Asuswrt-Merlin/asuswrt-merlin.legacy #1427: RT-N66U Loss of connectivity noticed on wired LAN during port scan Firmware:380.67

GitCartel, 2017-07-22:
GRC shields-up totally kills the router temporarily.
I thought DDOS protection is supposed to protect from denial of service, not create denial of service.
This is no good, I've owned dozens of routers from the DI-524 to this RT66N-U and everything in between and in 15 years I've never had a router crap-out like this.
CPU goes to the roof and its frozen till the port scan stops.
I cant believe I drove 140KM today to buy this router and seeing this.

Unacceptable

RMerl, 2017-07-22:
Not a firmware issue.

closed 2017-07-22 by RMerl

krnkris, 2017-07-22:
Have RT66N-U + Merlin 380.66
No problem or issue with it.
ROUTER + VPN stable

@GitCartel 
Did you made hard reset?
Followed the instructions?

@RMerl 
Thank you for your hard work with the firmware.

GitCartel, 2017-08-17:
not a firmware issue?
how come it doesn't do this using tomato?
you guys dont back your own work, just who cares attitude when a persons security is at risk.

thanks alot.

krnkris, 2017-08-17:
@GitCartel 

I just tested - GRC shields-up - within 10seconds went trough.
All is green. No risk.

Using Linux OS, RT66N-U + Merlin 380.66.

In case of security you should concern to use different OS.

Hope this helps.

RMerl, 2017-08-17:
This is a bug tracker, not a support site.  Your problem is a configuration issue - you most likely enabled logging of dropped packet, which combined with a port scan resulted in you DoS'ing yourself.  Had you gone to the support forums, you would have seen this already posted there.