Asuswrt-Merlin/asuswrt-merlin.legacy #1428: NTP queries use ntp_server1 - not settable via GUI

cbz, 2017-07-23:
Due to the following code:

https://github.com/RMerl/asuswrt-merlin/blob/4620b188428fb2cc867f08699879390497247fd0/release/src/router/rc/ntp.c#L189

Both ntp_server1 and ntp_server0 are used as valid values for ntp servers by the NTP subsystem, and by default the router round robins between each of them ntp_server0 is provisionable via the GUI, ntp_server1 is not, and by default is set to time.nist.gov.

In my opinion the code stanza should be changed, or http://router.asus.com/Advanced_System_Content.asp should be changed to enable both values to be set.

RMerl, 2017-07-23:
This was a recent change by Asus.  I don't know if they intend to eventually expose both settings or not, I'm waiting to see where they are heading with this change first.

RMerl, 2017-08-06:
Addressed with commit c6c617b998aa448f396a2543704dbfb26e3f9924.

closed 2017-08-06 by RMerl