Asuswrt-Merlin/asuswrt-merlin.legacy #1433: RT-AC88U lost 5G WiFi

ReqPro, 2017-07-29:
I was wondering why, during a reboot, the 5ghz LED wasn't lighting up and I couldn't find neither the regular SSID or the guest SSID, so I checked the configuration to see what could cause the 5GHz to fail (picture 1).

[image]

It tells me I need to activate 5GHz, so I decided to check the 5GHz settings before checking if radio is enabled, as I haven't touched that setting (picture 2). There I saw that it is using channel 0.

[image]

From there I decided to check if the radio was enabled, just to be sure (picture 3) and confirmed that it was indeed enabled.

[image]

Lastly I checked the system log, to make sure that the WiFi interfaces are brought up correctly, and they seem to be (picture 4).

[image]

It does not help to reboot, restore factory settings, go to recovery or anything else I could think of.

themiron, 2017-07-29:
most likely it's hw issue. try official fw and apply for rma if doesn't help.

RMerl, 2017-08-06:
Please use the support forums for support requests.

closed 2017-08-06 by RMerl