Asuswrt-Merlin/asuswrt-merlin.legacy #1434: RT-AC68U: QoS up/down traffic is treated in reverse when using OpenVPN Client

carmanj, 2017-07-29:
Using firmware: 380.67

I first noticed this the other day when downloading a large file in my Transmission client. It appeared that in the QoS Bandwidth Monitor Gauges appeared to be reflecting that I was uploading a large amount of data, not downloading. Additionally, it appears as though it is actually applying the QoS rules in the same manner (the limited upload speed is actually affecting the download speed). All other clients/traffic that isn't having the VPN's redirect policy rule applied, appears to be correctly having QoS applied as well as displaying properly.
[image]

RMerl, 2017-07-31:
It's not swapped, it's something broken at the lower levels  Asus are aware of the issue, however it's not a priority for them to fix as they consider Adaptive QoS a better alternative.

RMerl, 2017-07-31:
Actually, it's a different (also known) issue since you are using a VPN client.

closed 2017-07-31 by RMerl