Asuswrt-Merlin/asuswrt-merlin.legacy #1435: sha256.sum contains ^M at the end of the line

DiWa51, 2017-07-31:
Hi,
Under linux, when I'm checking the integrity of a new package, `sha256 -c sha256.sum` always fails due to the `^M` EOL in this file:
```
$ sha256sum -c sha256sum.sha256 
sha256sum: 'RT-AC66U_380.67_0.trx'$'\r': No such file or directory
: FAILED open or read
sha256sum: WARNING: 1 listed file could not be read
```
I'm surprised given the `unix2dos` used while generating this file: `sha256sum $FWNAME | unix2dos > sha256sum.sha256`
It still occurred with the latest RT-AC66U_380.67_0.zip package.
Any hint?

moooooooo, 2017-07-31:
dos2unix will strip out those characters.

DiWa51, 2017-08-01:
I'm just figuring out that `unix2dos` is used instead of `dos2unix` in tools/build-all to generate the file!

Is this done on purpose?
Then, does it mean Windows and Linux worlds can't benefit from sha256sum.sha256 without further manipulation (_unix2dos_)?

Am I missing something?

Cheers

token0, 2017-08-01:
DiWa51, If I would save this file under windows, I would like to keep it working after such manipulation. dos2unix just before build process insures that this file behaves unix-way.

DiWa51, 2017-08-01:
Just seing the notification on the README.md:
> Please do not use Github's Issue tracker for support requests. Use the support forums at SNBForums instead

It seems other faced such issue:
https://www.snbforums.com/threads/release-asuswrt-merlin-380-67-is-now-available.40191/page-14#post-338637

closed 2017-08-01 by DiWa51