Asuswrt-Merlin/asuswrt-merlin.legacy #1439: Update to nano 2.8.6

feranick, 2017-08-02:
Changelog:

GNU nano 2.8.6 "Kekulé" offers a new feature: the ability to do softwrapping between words -- at whitespace -- instead of always at the edge of the screen.  This can be activated with -a or --atblanks or 'set atblanks' together with the softwrap option.  This release further fixes a handful of rare display glitches, fixes a build failure on AIX, harmonizes the quoting rules in the rc
files, and renames the option 'cut' to 'cutfromcursor' (please update your nanorc files before 2020).

RMerl, 2017-08-06:
Going to wait to be closed to 380.68's final development stages in case 2.8.7 comes out before then.

feranick, 2017-08-07:
Sounds good. I'll update this bug report in case that happens. Thanks.

On 8/6/17 2:49 AM, Eric Sauvageau wrote:

Going to wait to be closed to 380.68's final development stages in case 2.8.7 comes out before then.

closed 2017-08-16 by RMerl

feranick, 2017-08-28:
Not sure if too late for this dev cycles... version 2.8.7 is available with useful fixes with using to edit python code. Full changelog:

GNU nano 2.8.7 "Fragrance" fixes a lockup when tabs are wider than the screen, makes indenting + unindenting more predictable by retaining relative indentations, allows pasting (^U) at a prompt, allows triple quotes in Python to not be followed by a character, does not scroll three pages on a roll of the mouse wheel, binds Alt+Up and Alt+Down to findprevious and findnext, and fixes some hard-to-describe issues with softwrapping and boundary-crossing tabs.

RMerl, 2017-08-29:
I haven't begun the 380.69 development cycle yet, I'll see what's the latest at that time.

commit 704da863 (2017-08-16): nano: Updated to 2.8.6 (closes #1439)