Asuswrt-Merlin/asuswrt-merlin.legacy #1443: OpenVPN client and "block-outside-dns" not working

neoatomic, 2017-08-06:
I'm running into an issue with the builtin OpenVPN client.
The VPN provider provides me with a .ovpn file that contains the option: "block-outside-dns".
If i understand correctly this option is included in OpenVPN v2.3.9+. The current Asuswrt-RMerlin build 380.68 uses OpenVPN 2.4.3. So i would expect that "block-outside-dns" is recognized as a command/function, but sadly it doesn't.  ;(

The below error is thrown in the log when starting the connection, and then fails to setup the connection.

``` 
Jul 31 23:14:51 openvpn[27289]: Options error: Unrecognized option or missing or extra parameter(s) in config.ovpn:38: block-outside-dns (2.4.3)
Jul 31 23:14:51 openvpn[27289]: Use --help for more information.
Jul 31 23:14:51 syslog: VPN_LOG_ERROR: 530: Starting OpenVPN failed... 
``` 
```RT-AC87U build 380.68```

TheUntouchable, 2017-08-07:
Its working here without any problems! But I am doing that with push from the server/router.

neoatomic, 2017-08-07:
@TheUntouchable Thanks for testing.
So seems this is specific to the VPN client, as you confirmed that the VPN Server does except the command in Merlin.
I'm wondering if maybe a Windows/machine build version or some sort of header response is needed for use of the "block-outside-dns" option?

TheUntouchable, 2017-08-07:
I am using the normal openvpn windows client (installer) from here without any modification on the client:
https://openvpn.net/index.php/open-source/downloads.html

RMerl, 2017-08-12:
That option only exists for Windows clients.

closed 2017-08-12 by RMerl